Ultimaker/Cura #8622: Support interface on tree supports causes an excessive horizontal offset

Liger0, 2020-10-26:
**Application version**
4.8 beta

**Platform**
Win10 x64

**Printer**
Cr10S

**Reproduction steps**
1. Tree supports
2. Interface support

**Screenshot(s)**
https://user-images.githubusercontent.com/20385116/97107743-6f829b00-16c9-11eb-96e8-963d55e1d2b4.png

**Actual results**
The tip is horizontally too far from the piece to support

**Expected results**
The tip is near the supported area

**Project file**
It's the sent I sent to @Ghostkeeper for #8218 

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

Asterchades, 2020-10-27:
To carry on here as to not stray off the subject at hand in the other thread...

@Ghostkeeper 
> I've never really seen the point of using support interface with tree supports. Tree supports already generate sort of an interface by converting the branches to slim lines, which is less configurable but does fit the shape of tree support better. 

Previously it worked well to bypass the issue with trees not reaching the correct Z offset, while also offering a less chaotic, more even surface. I've found the trees tend to create very wild "platforms" with massively complex geometry and weird cut-outs against the model which the interface mostly filled - particularly with larger horizontal surfaces to support. You can reduce these issues by setting the branch distance equal to or higher than the branch diameter (the defaults are 1mm and 2mm respectively) but that introduces gaps into the support structure.

Though I will be honest in that I've not actually tried printing it like that. The preview has put me off of it because it looks like a geometric mess that won't provide very even results, so I opt for something a little more "tried and true" as these tend to be larger prints as well.

I've no idea of the design feasibility, but having the trees support the interface would instil me with more confidence by virtue of the more consistent top to the supports. But trying to accommodate for a minimum interface area if present likely makes that more effort than the payoff would justify.

Ghostkeeper, 2020-10-27:
The point is though that support interface never gets wider than the support itself. I think what you saw was Creality's default of using a support interface with a sparse grid-like pattern and no outline. Because the support interface had no outline, it printed only where the grid lines happen to coincide with such a tiny tree support branch tip, giving a bunch of sparse and broken-up line segments. This leads to a much weaker connection between support and mesh. And probably gives the need to use 0 Z distance in the first place.
Try printing without support interface some day. The branch tips are indeed a bit chaotic down lower, but where they connect with the surface is a regular pattern designed to support 45 degree skin lines as best as possible with very little contact area.

Asterchades, 2020-10-27:
I actually haven't used a Creality profile in a long time. Even before I started modifying my Ender-2 I found the FFF base printer gave better results (especially following Creawesome's integration in 4.1.0), and I've since built a profile on top of that (one which now does dynamically disable the interface). Though I can't speak for Liger0 of course.

However, I can't say I'm really seeing the regular pattern on surface contact. Consider the following from 4.8.0 beta:
[image]
That's the top layer of tree supports under a 20mm calibration cube (hence the branches up the sides for the X/Y lettering) lifted 100mm off the bed, using the default FFF profile (though obviously with the dimensions adjusted so a 100mm rise will fit in the print volume) set to Draft. To me that looks fairly chaotic, albeit with perhaps some kind of over-arching orientation (which is indeed perpendicular to the model layer immediately above), and I just don't trust it to work - especially given some of those bits are just floating over air.

Plus there's still 2 layers (0.4mm) between that and the underside of the cube, despite the default 0.1mm offset. Oddly enough enabling the interface doesn't change that. The Z offset has to be set to 0 in either case in order to reduce the gap to one layer (the expected outcome of the offset being rounded up to 0.2mm), which in turn feeds back into Liger0's earlier report.

I could be looking too much into it, however. Like I said I haven't actually tested such a print so it could be a case of perception over reality, and zoom on the preview making things look worse than they are. I'll check it out tomorrow - though perhaps only 10mm off the bed, and only a flat square.

Asterchades, 2020-10-28:
Back with observations from my testing.

[image]

(Left) The defaults are... not good. Printing a 2mm mostly-flat square (OK, it's a 20mm calibration cube at 10% Z), 10mm above the build plate, using the FFF Custom profile (unchanged save build volume and enabling tree supports). It works, but it's clearly sagged, one of the corners isn't fully formed, the opposite corner doesn't meet with the skin, and it has a pronounced taper up the sides. But it did come away easily.

(Centre) Printing with the same basic settings but enabling the interface and reducing the Z gap to 0 gave a much better result (only 5% Z this time to save time/material), having only unexpected issues in the corners (which isn't actually unexpected given the trees don't like touching corners, for some reason - that seems a significant issue to me given corners require support the most). It was significantly harder to remove but it still came away cleanly once it started.

(Right) Last setting was adjusting the branch distance to the branch diameter divided by √2, making it 1.414mm instead of the default 1mm (going to the same 2mm as the branch diameter left too many inconsistent gaps for my liking). Z gap left at 0. This again has a decent surface quality but the corners suffer due to lack of direct support, and once again was more awkward to start removing but came away cleanly after it started.

So, OK, the interface didn't really help much on its own. It was the reduction in Z offset that made the most difference, which means if that weird missing layer (due to peaks/cones, as I recall) can be addressed that part should be remedied. But obviously, as Liger0 mentioned in the other thread, having the interface is important in terms of soluble materials, which means the interface X/Y offset needs to be figured out.

The corners, though, are a problem. I've seen this with other prints, too, in that the branches tend towards the middle of whatever they're supporting, rather than the edges - especially the corners. This greatly reduces the overall effectiveness, though is a separate issue. Would you like me to make a new thread on that or just leave it with you?

Ghostkeeper, 2020-10-28:
I tested tree support originally with PVA as well but found that to not work very well due to all of the starts and stops. PVA by nature needs to keep flowing. But yes having too much Z distance is still the most major issue I know of in tree support.

Disregarding all of that, the original issue of this thread - support interface causing excessive horizontal offset - is still a bug.

GregValiant, 2023-09-16:
@Asterchades  Can we close this?  It's been a while and tree supports have been re-worked for the 5.x versions of Cura.

Asterchades, 2023-09-17:
I would imagine so. I still feel like the new trees have a lack of emphasis on corners but the primary issue of X/Y offset would appear to have been resolved.

Honestly a little surprised this wasn't flagged as "Stale" with the pass recently.

closed 2023-09-17 by Asterchades

GregValiant, 2023-09-18:
A surprising number of old ones weren't marked as stale.  I was going through and looking for requests and/or ideas for post processors so I flagged a few and sent some messages.