Ultimaker/Cura #8623: Application Startup Broken (Doesnt Start)

bryanhunwardsen, 2020-10-26:
**Application version**
4.7.0 & 4.7.1 (at least)

**Platform**
Windows 10 2004 and Insider  10.0.20241.1005

**Printer**
(Custom) Ender3V2

**Reproduction steps**
(Multiple Independent machines/installations all affected)
1) Start Cura early last week (Loads at its normal slow speed but it loads)
2) Start Cura late last week Cura startup hangs on loading settings (still at small splash screen)
2.1) Click on splash screen, windows reports Cura.exe not responding, begins windows crash flow.
2.2) Chask task manager for cura.exe process/ram/network/file system images, no indication of issue all values very low for exe and system
2.3) Check log file, no errors in the tail to indicate the issue, few warnings about inability to load unnamed plugin.
2.4) Force close process
2.5 repeat several times, no change in behavior across reboots, task kills, upgrade to 4.7.1, etc
As I had not changed anything my expectation was that some remote/network integration is failing, or a settings config was corrupted and Cura hangs unable to log and recover from such failure.

I tried 4 days later hoping any remote depedency that had broken might be repaired but same result.
And then I walked away and came back 30 minutes later to find Cura loaded(!!!) which seems to indicate my profiles/settings are still valid.
Trying to repro I found that Cura hangs for 20 mintutes between the 2 below log lines, with no indication why, but all loading before and after the hang proceeds as normal.

**This happens on multiple machines with vastly different hardware installed that all worked just fine this time last week but now all suffer this issue with no changes on my part.

**Screenshot(s)**
N/A

**Actual results**
App startup hangs for 20 minutes

**Expected results**
No delay should occur

**Project file**
N/A

**Log file**
(TLDR)
2020-10-26 15:39:25,034 - DEBUG - [MainThread] UM.PluginRegistry.loadPlugins [373]: Loading all plugins took 2.7668869495391846 seconds
2020-10-26 15:39:25,049 - INFO - [MainThread] UM.VersionUpgradeManager.upgrade [163]: Looking for old configuration files to upgrade.
**2020-10-26 15:39:25,400 - INFO - [MainThread] UM.VersionUpgradeManager.upgrade [175]: Checking and performing updates took 0.35339903831481934
2020-10-26 15:58:18,186 - DEBUG - [MainThread] UM.Settings.ContainerRegistry.loadAllMetadata [361]: Loading metadata into container registry took 1.0665972232818604 seconds**
2020-10-26 15:58:18,203 - DEBUG - [MainThread] cura.Scene.CuraSceneController.setActiveBuildPlate [142]: Select build plate: 0

20 minutes bewtween log lines with no error or warning

(FULL)
will provided if needed

**Additional information**
repro'd on fully loaded core7/intel grfx Lenovo X1 Yoga and custom Desktop AMD Threadripper / Radeon 7 grfx fast and lots of ram/pcie ssd.

bryanhunwardsen, 2020-10-26:
Here is a longer log snippet from my desktop, the delay is still there but closer to 5 minutes which may reflect the compute differential compared to the OP log provided from my laptop.

2020-10-23 17:32:07,385 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin FileLogger
2020-10-23 17:32:07,404 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin FirmwareUpdateChecker
2020-10-23 17:32:07,418 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin FirmwareUpdater
2020-10-23 17:32:07,429 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin GCodeGzReader
2020-10-23 17:32:07,442 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin GCodeGzWriter
2020-10-23 17:32:07,455 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin GCodeProfileReader
2020-10-23 17:32:07,481 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin GCodeReader
2020-10-23 17:32:07,494 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin GCodeWriter
2020-10-23 17:32:07,513 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin ImageReader
2020-10-23 17:32:07,526 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin LegacyProfileReader
2020-10-23 17:32:07,539 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin LocalContainerProvider
2020-10-23 17:32:07,556 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin LocalFileOutputDevice
2020-10-23 17:32:07,570 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin MachineSettingsAction
2020-10-23 17:32:07,584 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin ModelChecker
2020-10-23 17:32:07,597 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin MonitorStage
2020-10-23 17:32:07,613 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin PerObjectSettingsTool
2020-10-23 17:32:07,629 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin PostProcessingPlugin
2020-10-23 17:32:07,641 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin PrepareStage
2020-10-23 17:32:07,652 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin PreviewStage
2020-10-23 17:32:07,670 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin RemovableDriveOutputDevice
2020-10-23 17:32:07,681 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin SentryLogger
2020-10-23 17:32:07,711 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin SimulationView
2020-10-23 17:32:07,728 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin SliceInfoPlugin
2020-10-23 17:32:07,745 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin SolidView
2020-10-23 17:32:07,761 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin SupportEraser
2020-10-23 17:32:07,809 - WARNING - [MainThread] UM.Logger.logException [106]: Exception: Failed to load packages for Marketplace
2020-10-23 17:32:07,813 - WARNING - [MainThread] UM.Logger.logException [110]: NoneType: None
2020-10-23 17:32:07,817 - WARNING - [MainThread] UM.Logger.logException [106]: Exception: Failed to load packages for Marketplace
2020-10-23 17:32:07,821 - WARNING - [MainThread] UM.Logger.logException [110]: NoneType: None
2020-10-23 17:32:07,825 - WARNING - [MainThread] UM.Logger.logException [106]: Exception: Failed to load packages for Marketplace
2020-10-23 17:32:07,829 - WARNING - [MainThread] UM.Logger.logException [110]: NoneType: None
2020-10-23 17:32:07,832 - WARNING - [MainThread] UM.Logger.logException [106]: Exception: Failed to load packages for Marketplace
2020-10-23 17:32:07,836 - WARNING - [MainThread] UM.Logger.logException [110]: NoneType: None
2020-10-23 17:32:07,848 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin Toolbox
2020-10-23 17:32:07,863 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin CameraTool
2020-10-23 17:32:07,881 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin MirrorTool
2020-10-23 17:32:07,903 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin RotateTool
2020-10-23 17:32:07,922 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin ScaleTool
2020-10-23 17:32:07,937 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin SelectionTool
2020-10-23 17:32:07,957 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin TranslateTool
2020-10-23 17:32:07,970 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin TrimeshReader
2020-10-23 17:32:07,991 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin UFPReader
2020-10-23 17:32:08,009 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin UFPWriter
2020-10-23 17:32:08,025 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin UltimakerMachineActions
2020-10-23 17:32:08,208 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin UM3NetworkPrinting
2020-10-23 17:32:08,223 - INFO - [Thread-1] UpdateChecker.UpdateCheckerJob.run [40]: Checking for new version of cura
2020-10-23 17:32:08,229 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin UpdateChecker
2020-10-23 17:32:08,720 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin USBPrinting
2020-10-23 17:32:08,938 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin VersionUpgrade21to22
2020-10-23 17:32:09,004 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin VersionUpgrade22to24
2020-10-23 17:32:09,070 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin VersionUpgrade25to26
2020-10-23 17:32:09,136 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin VersionUpgrade26to27
[... cut ...]
2020-10-23 18:52:33,845 - DEBUG - [MainThread] cura.AutoSave._onTimeout [59]: Autosaving preferences, instances and profiles
2020-10-23 19:01:27,411 - DEBUG - [MainThread] cura.OAuth2.AuthorizationHelpers.getAccessTokenUsingRefreshToken [60]: Refreshing the access token.
2020-10-23 19:01:28,243 - DEBUG - [MainThread] cura.OAuth2.AuthorizationService._storeAuthData [250]: Attempting to store the auth data
2020-10-23 19:01:38,238 - DEBUG - [MainThread] cura.AutoSave._onTimeout [59]: Autosaving preferences, instances and profiles

fieldOfView, 2020-10-27:
Do you perhaps have a drive mapped to the letter X that is not currently connected? If so, could you try mapping that drive to a different driveletter?

bryanhunwardsen, 2020-10-27:
@fieldOfView On my Desktop I have a network drive mapped to "S:" and nothing on my laptop, neither has any mapping or drives assigned to "X:"

Ghostkeeper, 2020-10-28:
We can rule out the version upgrade and the loading of the metadata here. Both are timed almost completely. These are the things that are done in between, in the start-up sequence:
* Loading preferences file (the splash window also shows "loading preferences..." then)
* Check if you are running a new version of Cura for the first time (so change log should be shown then).
* Filling list of recent files.
* Adding system tray icon.
* Loading translation catalogue for Cura.
* Connect some signals for when the scene gets updated. One of these is on a 0.5s delay which may have had a reason that could be defeated by slow performance.
* At this point the splash screen says "loading machines...".
* Obtain the lock file to prepare loading the profile metadata.

One thing that could feasibly take a long time is adding the system tray icon, since that is kind of dependent on the operating system which is at liberty to block such things. This is also a visible part of the process. Perhaps you can start Cura to see if the system tray icon is there? If it is, the slow part must be somewhere after that in the process. If there is no system tray icon, the slow part was before that or the system tray icon itself.

Another thing that would help is to have more logging in our code regarding this. There are quite a few steps here that don't get a log entry. They should though...

bryanhunwardsen, 2020-10-28:
@Ghostkeeper I tried to re-run to check tray icon, alas it is now loading correctly.
Neither machine has been rebooted, or been used for much else other than testing this Cura bug during this time, which again leads me to believe there is a remote resource disturbance that caused the issue. For a bug like this to replicate so perfectly across machines, break at the same time, and resolve identically, with no changes on my end, typically points to either a remote resource or some type of datetime bug.

I guess this can technically be closed now, but, having Cura magically break and become unusable for a week, then magically fix itself, is problematic at best.  I would suggest if this can somehow get in someone's backlog, to add the additional logging you indicated above, otherwise it will be near impossible to troubleshoot this if it crops up again.

Thanks for the support
humblest regards-

closed 2020-10-28 by bryanhunwardsen