Ultimaker/Cura #8624: [4.8 beta] Inner perimeters are printed overlapped, and distant from outer wall

Liger0, 2020-10-27:
**Application version**
4.8 beta

**Platform**
Windows 10 x64

**Printer**
Cr10S

**Reproduction steps**
1. at least 4 total walls

**Screenshot(s)**
[image]

**Actual results**
The inner perimeter looks like a single line, but it is made up of 2 lines instead.
The outer most green line is extremely small, printed on the most inner one itself (nozzle will crash on the printed surface), and it lefts out a lot of space from the outer wall.
Also, the small line is weirdly segmented.

**Expected results**
Nicely adjacent lines

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save Project. For big files you may need to use WeTransfer or similar file sharing sites. G-code files are not project files!)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
I know a better gap filling been worked on. Is there actually an option I can't find to set the minimum and maximum extrusion flow (thickness) for gap filling and inner wall lines?

printingotb, 2020-10-27:
I believe this is expected behaviour when using "compensate inner wall overlaps".
[image]

Ghostkeeper, 2020-10-28:
It's a bit hard to see in those small thumbnails, but the screenshots are different. You can see it better in the article itself: https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/shell/travel_compensate_overlapping_walls_x_enabled.md

This is indeed expected behaviour, since the compensation for wall overlap doesn't actually change the positions of the walls. It just changes their width. They will not automatically re-center to be in the middle between other lines.
In practice, if you actually print this, the nozzle opening will slide partially over the previously printed lines while extruding exactly enough material to fill in the gap slightly to one side of the opening. The pressure from the nozzle then squeezes the material to that side to fill the gap. So I'd expect it to come out all right in the final print.

A better solution is being created in Ultimaker/CuraEngine#1210 . In that branch, the wall overlap compensation has been removed completely in favour of having variable line width.

closed 2020-10-28 by Ghostkeeper

Liger0, 2020-10-28:
> 
> A better solution is being created in [Ultimaker/CuraEngine#1210](https://github.com/Ultimaker/CuraEngine/pull/1210) . In that branch, the wall overlap compensation has been removed completely in favour of having variable line width.

Will those features have also options with separate flow percentage? It would be usful to have a flow setting different for inner perimeters (green) and gap filling(yellow), different from the external and internal perimeters, which can be calculated precisely to make good fittings.

Ghostkeeper, 2020-10-28:
The Flow settings are still there, unchanged from their current behaviour.
There is no Flow setting for gap filling though. There never was.

Liger0, 2020-10-28:
> 
> 
> The Flow settings are still there, unchanged from their current behaviour.
> There is no Flow setting for gap filling though. There never was.

That is what I would like in the new implementation indeed, a separate flow for internal walls and gap filling..