Ultimaker/Cura #8626: Gapfill appears on the extra skin wall of expanded skin. 4.8 beta

printingotb, 2020-10-27:
**Application version**
4.8 beta

**Platform**
Windows 7, NVIDIA GeForce GT 730. (Using dual monitors).

**Printer**
Custom fff

**Reproduction steps**
1. See attached project. Fill gaps is active, Extra skin wall is on.
2. Turning off fill gaps removes the short tiny lines. 
    Surprisingly, setting infill density to 0 also removes them.

**Screenshot(s)**
[image]

**Actual results**
Gap-fill at edge of extra skin wall

**Expected results**
No gap-fill

**Project file**
[Gap fill at edge of skin expansion.zip](https://github.com/Ultimaker/Cura/files/5444447/Gap.fill.at.edge.of.skin.expansion.zip)

jellespijker, 2020-10-27:
Devs see CURA-7804. Same problem as reported at the eCCB

jellespijker, 2020-11-06:
Fixed for 4.8
Thank you for reporting it.

closed 2020-11-06 by jellespijker

Ghostkeeper, 2021-07-21:
4.9 and 4.10 had this bug again due to mistakes in merging, but for 4.11 it should be fixed again.