Ultimaker/Cura #8628: [4.8 beta] Infill wall lines overlaps when infill steps is active at teh same time

printingotb, 2020-10-27:
**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save Project. For big files you may need to use WeTransfer or similar file sharing sites. G-code files are not project files!)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

printingotb, 2020-10-27:
Submitted before I was done, updating...

jellespijker, 2020-10-27:
This is occurs because the gradual infill areas don't take into account previous generated area. This is normally not a problem but once _connected lines_ are enabled or _extra infill wall counts_ are set to a number higher then 0.  It will generate an infill area with a denser density with an extra wall around it. Since these denser areas are drawn over the a previous generated infill area this will result in over extrusion. 

We were aware of this problem with _connected lines_ (Cura devs see CURA-7115) but not with _extra infill wall count_. This is fortunately also fixed in branch  [CURA-7115_gradual_infill_overextrusion](https://github.com/Ultimaker/CuraEngine/tree/CURA-7115_gradual_infill_overextrusion).
Were we only generate a wall for the least dense area 
[FffGcodeWriter.cpp#L1708](https://github.com/Ultimaker/CuraEngine/blob/d759e19df7d454032aabe8205bee594e032e39a7/src/FffGcodeWriter.cpp#L1708)

closed 2020-11-06 by rburema