Ultimaker/Cura #8629: Modify Settings per Overlap that the Quality/Layer Height be included!

Asho777, 2020-10-27:
I would like to suggest that Layer Height be added to the Quality settings, while in the Modify Settings per Overlap.
This will allow for Layer Heights to be changed at manually controlled heights for the selected model!
[image]

**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

nallath, 2020-10-27:
I agree that it would be amazing! Unfortunately, it's a technical limitation that we don't allow it.

The main reason for this is that with modifier meshes, you can also select a part of the model. This would mean that you could have multiple layerheights at any given height. That would in turn mean that you would need to calculate what areas to avoid and in what sequence these should be printed.

I don't think that we will be able to implement this any time soon, as it's insanely difficult to do.

closed 2020-10-27 by nallath

Ghostkeeper, 2020-10-28:
See also https://github.com/Ultimaker/Cura/issues/3193
Layer height is a particular difficulty due to collision checking with other layers.

Asho777, 2020-10-29:
Yes, it would be amazing to have multiple layer heights at a given layer height, nallath. But I would only expect this setting to be across the entire layer, not across selected areas of the layer! The end result would be exactly the same as Adaptive Layers do now, except that you would have more manual control over the areas of the print that would change layer height! Maybe the blocker and Modify Settings per Overlap, is not the correct area for this change. Maybe in the Post Processing Script could be added to allow for a layer height change at a height and or layer number! That is the end result I was requesting!

Ghostkeeper, 2020-10-30:
What do you think would happen if you have a layer height of 0.1mm in the setting, but two equal-sized objects in the scene side by side which override the layer height to 0.06mm and 0.15mm?

I think a better interface for this would be to indicate regions on the layer slider, sort of per-layer settings. I think Simplify3D had something like that.

Asho777, 2020-10-30:
Yes I agree that it would be very difficult to have two objects printing at different layer heights. I think my suggestion may have been incorrectly understood. A manual layer change as of a layer number or dimensional height, would be across all objects on the build plate. So if you have five (5) different objects (models) laid out on the build plate all objects would have the layer height changed!
Yes I do agree with your interface idea, to have regions or levels along the layer slider to select the layer heights. These levels selected on the slider would also be across every object in the total build volume!
S3D has processors, for which you can assign to any object and all objects on the build plate. The only reason I use S3D is because of the processors, and the ability to be able to change layer heights and any layer level. I also used S3D when print model aircraft where the models have only 1 wall or parameter, except 1mm up from the bed. But I can do this in Cura now, but not the layer height change!

Ghostkeeper, 2020-11-03:
I think you'll be interested in this issue then: https://github.com/Ultimaker/Cura/issues/7488