Ultimaker/Cura #8630: [4.8 beta] Settings field loses the scroll bar while categories remain outside the field.

printingotb, 2020-10-27:
**Application version**
at least from 4.3 to 4.8 beta

**Platform**
Windows 7, NVIDIA GeForce GT 730. (Using dual monitors).

**Printer**
Custom fff

**Reproduction steps**
This has happened to me occasionally for quite some time but I have so far not been able to reproduce it, until yesterday.
But today I didn't succeed so there must be something more going on.
This is what I did yesterday:
1. Expand setting categories, close categories starting with the bottom ones and work yourself upwards. 
2. When closing the Quality category, the scroll bar disappears but the quality category is outside the window. The scroll wheel doesn't work either. Expanding any category will restore the scroll bar. Minimizing the window will also restore it.

**Screenshot(s)**
[image]

**Actual results**
Category outside the field

**Expected results**
Quality should be at he top, not outside.

Liger0, 2020-10-27:
#6211

tlhintoq, 2020-10-27:
I've seen it as well.  It isn't reproducable in a step-by-step.  Pretty sure its a threading issue so you have to hit it at just the right time 1 in 20.
Most often I've seen it when a mix of categories are open/closed then do a search for a term, then clear the search term.  There's so many changes that take place at once with closing categories that were only open temporarily that I think the events stomp on each other and the UI gets lost.

Ghostkeeper, 2020-10-28:
Indeed this is a duplicate of #6211. Please continue the conversation there.
It seems to be semi-random and perhaps dependent on the Qt version.

closed 2020-10-28 by Ghostkeeper