Ultimaker/Cura #8631: [4.8 beta] Crash when slicing

Liger0, 2020-10-27:
**Application version**
4.8 beta

**Platform**
Windows 10 x64

**Printer**
Cr10S

**Reproduction steps**
1. Imported a model, rotated, sliced

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
Crash report from windows saying "Slicing crash".
Cura didn't freeze, it kept on and "working", but the slicing bar was empty and not moving.
I can't reproduce it.

**Expected results**
No crash

**Project file**
[CCR10S_slicing crash.zip](https://github.com/Ultimaker/Cura/files/5445227/CCR10S_slicing.crash.zip)

**Log file**
[cura.zip](https://github.com/Ultimaker/Cura/files/5445228/cura.zip)

**Additional information**
(Extra information relevant to the issue.)

konskarm, 2020-10-27:
I cannot reproduce the crash. It happens sometimes that CuraEngine may indeed crash and lead to that issue, but in this case it seems we don't have reliable reproducton steps as this probably was a one-time thing, right? 
If you manage to reproduce it reliably, please post the reproduction steps.

Liger0, 2020-10-27:
It was a one time thing.

konskarm, 2020-10-27:
I'll close this then. Feel free to open another ticket if you manage to find good reproduction steps.

closed 2020-10-27 by konskarm