Ultimaker/Cura #8632: [4.8.0] Fails to import profiles correctly from 4.7.1

tlhintoq, 2020-10-27:
Image says it all.
4.7.1 has no issues side-by-side with 4.8.0 which is complaining about the profiles being corrupt.
If the user can't upgrade without getting nuked back to the stone age, they don't want to upgrade.
[image]

nallath, 2020-10-27:
Did you perhaps manually add profiles? If so, that might be the cause for it.

closed 2020-11-10 by no-response[bot]

bcpusscat, 2020-11-11:
I have a similar problem, profiles from 4.7.1 are imported ok but are not displayed
I get this message:

The profile is not visible because its quality type 'normal'
is not available for the current configuration
switch to a material/nozzle combination that can use this quality type.

I am using Filamentum carbon CF15 which is ok in 4.7.1. for the moment I am uninstalling 4.8.0 as I cannot use it.

bcpusscat, 2020-11-11:
Also installing 4.8.0 has screwed the profiles from 4.7.1

Ghostkeeper, 2020-11-17:
Which profile did you try to import on which printer, @bcpusscat?

4.8 should never edit 4.7.1's profiles at all. That is a very serious issue and you'd do best to report that in a new issue (with properly filled in reproduction steps). Unless you happen to accidentally choose to remove the 4.7 profile when uninstalling 4.7, which is somewhat of a common problem if you didn't read that pop-up. I haven't seen that issue come up myself, nor our testers.

bcpusscat, 2020-11-18:
Ghostkeeper.

I didn't uninstall 4.7 when I installed 4.8. I exported the profiles from
4.7 for import in 4.8
It looked like they imported ok but they would not display in the profile
list, see error message below

The first error says, "Successfully imported profile......   Warning: the
profile is not visible its quality type 'normal' is not available for the
current configuration.
Switch to a material/nozzle combination that can use this quality type.

I had to recreate the profiles in 4.8.

Regards,

Phil

-- 

Philip J. Sharrock

Development Support Engineer.

[image: 4Sight Imaging]

t:
f:
w:
a:

+44 1204 216162
+44 1204 216167
www.4sightimaging.com  <http://www.4sightimaging.com/>
4Sight Imaging Ltd, 5th Floor Left, 120 Bark Street, Bolton, BL1 2AX
<https://maps.google.com/?q=120+Bark+Street,+Bolton,+BL1+2AX&entry=gmail&source=g>

* This e-mail and its attachment(s) has been sent from and on behalf of
4Sight Imaging Ltd (Registered in England 05214775), whose registered
office is at PM+M, Greenbank Technology Park, Challenge Way, Blackburn, BB1
5QB. *

* The contents of this e-mail and its attachment(s) are confidential to the
intended recipient. It may not be disclosed, copied, forwarded, used or
relied upon by any person other than the intended addressee.  If you
believe that you have received the e-mail and its attachment(s) in error,
you must not take any action based on them, nor must you copy or show them
to anyone. Please respond to the sender and delete this e-mail and its
attachment(s) from your system. *

Ghostkeeper, 2020-11-20:
I mean, that message is supposed to show up if the quality level of the profile is not available for the current configuration. Hence why I'm basically asking which configuration you were using, and which profile (so I can see what quality level you were trying to import for).

bcpusscat, 2020-11-20:
Problem has been sorted, I recreated the profiles in 4.8 then deleted the
4.7 profiles

The quality level was 'normal' with Filamentum CF15 0.4 nozzle

Regards,

Phil

-- 

Philip J. Sharrock

Development Support Engineer.

[image: 4Sight Imaging]

t:
f:
w:
a:

+44 1204 216162
+44 1204 216167
www.4sightimaging.com  <http://www.4sightimaging.com/>
4Sight Imaging Ltd, 5th Floor Left, 120 Bark Street, Bolton, BL1 2AX
<https://maps.google.com/?q=120+Bark+Street,+Bolton,+BL1+2AX&entry=gmail&source=g>

* This e-mail and its attachment(s) has been sent from and on behalf of
4Sight Imaging Ltd (Registered in England 05214775), whose registered
office is at PM+M, Greenbank Technology Park, Challenge Way, Blackburn, BB1
5QB. *

* The contents of this e-mail and its attachment(s) are confidential to the
intended recipient. It may not be disclosed, copied, forwarded, used or
relied upon by any person other than the intended addressee.  If you
believe that you have received the e-mail and its attachment(s) in error,
you must not take any action based on them, nor must you copy or show them
to anyone. Please respond to the sender and delete this e-mail and its
attachment(s) from your system. *