Ultimaker/Cura #8635: Cura 4.7.1 - 3 bugs

Silithas, 2020-10-27:
**Application version**
4.7.1

**Platform**
Windows 10 1909

**Printer**
Ender 3 pro with skr 1.3

Issue 1: text doesn't appear on top of icons for support blocker:

**Reproduction steps**
1. Hover over support blocker, but neither it or the icon above shows text explaining what each does (as occasionally they switch place and i don't ever remember what support blocker looked like)
2. (Something you did next.)

**Screenshot(s)**
https://i.imgur.com/OCILTHr.gif

**Actual results**
Nothing happens, they change shade highlighting i have my mouse over it, but no text appears

**Expected results**
Text should appear telling me what is what.
________________________________________________

Issue 2: Support blocker doesn't block supports

**Reproduction steps**
1: choose support blocker, place it where support is, click slice, but support is still there.

**Screenshot(s)**
https://i.imgur.com/DlL1bRI.png

**Actual results**
Cube is still there, but so is the supports.

**Expected results**
Supports should be gone from the area chosen.

**Project file**
https://www.dropbox.com/s/bpqkukdq9nc5lcn/issue%20project%20file.3mf?dl=0

**Log file**
https://www.dropbox.com/s/gsdzzczwgtnsd3i/cura.zip?dl=0

____________________________________________________

Issue 3: Resize arrow cubes does the opposite of direction dragged

**Reproduction steps**

Move the red resize arrow outwards, and the red arrow's direction moves instead. Same when moving red, then green's direction moves

**Screenshot(s)**
https://i.imgur.com/WcBWpx0.gif

**Additional information**
(Extra information relevant to the issue.)

nallath, 2020-10-27:
Issue 1: Confirmed. This used to work in 4.6, it doesn't in 4.7 / 4.8

Issue 2: You added a "support mesh". This is not the same as a support blocker. A support mesh is to indicate that you want that object to be printed as if it's support (so in that sense it's almost the inverse)

Issue 3: Fixed in the 4.8 Beta.

Silithas, 2020-10-27:
> Could you for the next time create 1 issue for each bug that you have? It makes it a lot easier for us to keep track of things.
> 
> Issue 1: Confirmed. This used to work in 4.6, it doesn't in 4.7 / 4.8
> 
> Issue 2: You added a "support mesh". This is not the same as a support blocker. A support mesh is to indicate that you want that object to be printed as if it's support (so in that sense it's almost the inverse)
> 
> Issue 3: Fixed in the 4.8 Beta.

Will do.

Regarding issue 2: I tried to use both, but as i don't remember which was the blocker, so i tried both, and the blocker didn't block supports in 4.7-4.8

Ghostkeeper, 2020-10-28:
This is what I see when I change your project file to make it a support blocker:
[image]

And this is what I see when I remove the support blocker:
[image]

So the support seems to be blocked correctly. Note that there is still a layer of support at the bottom there, because there is a tiny ledge just beneath the support blocker that still gets support, and this support is joined to the bigger piece of support due to Support Join Distance.

Issue 1 is a duplicate of https://github.com/Ultimaker/Cura/issues/8324.
Issue 3 is a duplicate of https://github.com/Ultimaker/Cura/issues/4136, https://github.com/Ultimaker/Cura/issues/3625, https://github.com/Ultimaker/Cura/issues/4309, https://github.com/Ultimaker/Cura/issues/4386, https://github.com/Ultimaker/Cura/issues/5221, https://github.com/Ultimaker/Cura/issues/5189, https://github.com/Ultimaker/Cura/issues/5310, https://github.com/Ultimaker/Cura/issues/5462, https://github.com/Ultimaker/Cura/issues/5599, https://github.com/Ultimaker/Cura/issues/5602, https://github.com/Ultimaker/Cura/issues/5603, https://github.com/Ultimaker/Cura/issues/5872, https://github.com/Ultimaker/Cura/issues/5916, https://github.com/Ultimaker/Cura/issues/5965, https://github.com/Ultimaker/Cura/issues/6148, https://github.com/Ultimaker/Cura/issues/6150, https://github.com/Ultimaker/Cura/issues/6238, https://github.com/Ultimaker/Cura/issues/6370, https://github.com/Ultimaker/Cura/issues/6619, https://github.com/Ultimaker/Cura/issues/6770, https://github.com/Ultimaker/Cura/issues/6829, https://github.com/Ultimaker/Cura/issues/6890, https://github.com/Ultimaker/Cura/issues/6971, https://github.com/Ultimaker/Cura/issues/7102, https://github.com/Ultimaker/Cura/issues/7142, https://github.com/Ultimaker/Cura/issues/7237, https://github.com/Ultimaker/Cura/issues/7388, https://github.com/Ultimaker/Cura/issues/7446, https://github.com/Ultimaker/Cura/issues/7448, https://github.com/Ultimaker/Cura/issues/7582, https://github.com/Ultimaker/Cura/issues/7638, https://github.com/Ultimaker/Cura/issues/7703, https://github.com/Ultimaker/Cura/issues/7768, https://github.com/Ultimaker/Cura/issues/7778, https://github.com/Ultimaker/Cura/issues/7799, https://github.com/Ultimaker/Cura/issues/7946, https://github.com/Ultimaker/Cura/issues/8100, https://github.com/Ultimaker/Cura/issues/8122, https://github.com/Ultimaker/Cura/issues/8263, https://github.com/Ultimaker/Cura/issues/8304, https://github.com/Ultimaker/Cura/issues/8309, https://github.com/Ultimaker/Cura/issues/8339, https://github.com/Ultimaker/Cura/issues/8369, https://github.com/Ultimaker/Cura/issues/8392, https://github.com/Ultimaker/Cura/issues/8414 and https://github.com/Ultimaker/Cura/issues/8635.
So I'll change the title of this issue to only include issue 2.

Could you provide a project file that reproduces the issue?

voltmeter101, 2020-11-14:
Silithas, can you please send me an email to dmr6064@gmail.com

I noticed about another bug you were having and found a fix for it and wanted to ask for your help if i may because i am having the same problem
Sorry about hijacking the thread but i couldn't find any other way of reaching you on here. I used to be able to send messages i thought

nallath, 2021-05-11:
> This issue still exists. Some ppl in the forums have also described it: https://community.ultimaker.com/topic/27196-support-blocker-has-no-effect/
> 
> Would be good to disable this nonsensical bot.

I disagree. We put the label on it if we request more information (which we often do if we can't reproduce the issue ourselves). If we don't get more information, there is nothing we can do to resolve the situation.

Once someone provides a way for us to reproduce the issue, we would be more than happy to reopen it.

hasufell, 2021-05-11:
[cura_dsa.3mf.zip](https://github.com/Ultimaker/Cura/files/6458796/cura_dsa.3mf.zip)

Ghostkeeper, 2021-05-18:
@hasufell The support blocker does block support in your project file. However there are two reasons why the support is closed towards the bottom:

* There is a gap of less than the Support Join Distance in the support. Support Join Distance is set to 2mm, while the support blocker is less than 1mm wide. So any openings that this support blocker would create are going to be closed.
* It's generating a [support tower](https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/support/support_use_towers.md) because the supported piece is less wide than the [Maximum Tower-Supported Diameter](https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/support/support_tower_maximum_supported_diameter.md) I must say that the Maximum Tower-Supported Diameter setting of 3mm is quite high (but the contributor who created the profile there found it necessary apparently; maybe the printer is a bit wobbly). The support tower becomes wider at the bottom, which also closes the gap.

Disabling Use Towers and reducing Support Join Distance to 0.45mm or less keeps the gap open.

The support towers are also causing an old bug to appear here, leading to a quite useless thin pillar of support. That is a manifestation of this bug: https://github.com/Ultimaker/Cura/issues/7487 Since the support blocker is most likely intended to remove that pillar, I'll close this as a duplicate of that.

closed 2021-05-18 by Ghostkeeper