Ultimaker/Cura #8636: Feature Request: Retract + Wipe + ZHop AllTogether

GasAlbert, 2020-10-27:
Feature Request:

On traveling, doing a mix of already existing features could improve results:

Retract - to discharge pressure
Wipe - to avoid stringin
Z Hop - to mitigate oozing

I think those together could be a good combination. Ill try doing it myself, but leave it here too in case someone else could do it faster and better

Ghostkeeper, 2020-10-28:
Isn't it doing this already, if Retraction and Z Hop are enabled?

For wiping there is no specific setting. It'll always wipe when it needs to do a retraction while on a wall. For skin and infill there is a wipe setting. For support there is not.

GasAlbert, 2020-10-28:
Not exactly. When ZHop is enabled, it rises over the print and then travel out. On my printers this leaves strings all over since the PETg sticks to the surface.

Clarification: what I mean is to make a wipe leaving the printed area and after that rise the nozzle so if its a long travel and oozes, this excess material get squeeshed over the print, inside of it, and not hanging out from the skin. Trying different combinations of parameters could not make this behaviour.

Ghostkeeper, 2020-10-30:
I could. Here is a video that demonstrate the wiping behaviour just before it makes a Z hop: http://dulek.net/work/wipe_before_zhop.webm
It makes a travel move to the infill before it starts the Z hop. Then it retracts while it's in the infill area. Then it hops. It also lands inside the infill area (and in my case it starts with the infill in the next object so there is no wipe there). There is no setting to disable that or anything, it'll always do that as long as there is infill area nearby. Note that the hop is not visible in layer view unless you save the g-code and load it back in.

Did you mean that it needs to wipe somewhere else than the infill area?

closed 2020-11-13 by no-response[bot]