Ultimaker/Cura #8639: 4.8 Keeps resetting after it's closed.

oralallen82, 2020-10-27:
**Application version** 
4.8

**Platform**
OSX Catalina
Version 10.15.7

**Printer**
Creality CR6-SE
Creality Ender 3 Pro

**Reproduction steps**
1. Go through initial setup slice files etc.
2. Create a custom profile for each printer
3. Quit the app once complete
4. Reopen the app I am presented with the setup again
5. Complete the setup again. The profiles are still present
6. Close and reopen the issue occurs again.

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
Application resets

**Expected results**
Application to reopen without having to go through the setup process again.

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save Project. For big files you may need to use WeTransfer or similar file sharing sites. G-code files are not project files!)

**Log file**
2020-10-27 18:10:19,677 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin FileLogger
2020-10-27 18:10:19,682 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin LegacyProfileReader
2020-10-27 18:10:19,685 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin AMFReader
2020-10-27 18:10:19,688 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin SimpleView
2020-10-27 18:10:19,692 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin SolidView
2020-10-27 18:10:19,697 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin FirmwareUpdateChecker
2020-10-27 18:10:19,701 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin SentryLogger
2020-10-27 18:10:19,705 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin SupportEraser
2020-10-27 18:10:19,708 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin PrepareStage
2020-10-27 18:10:19,720 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin OrientationPlugin
2020-10-27 18:10:19,759 - WARNING - [MainThread] OctoPrintPlugin.OctoPrintOutputDevicePlugin.<module> [49]: Supplied version of Zeroconf module imported
2020-10-27 18:10:19,771 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin OctoPrintPlugin
2020-10-27 18:10:19,811 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin ThingiBrowser
2020-10-27 18:10:19,821 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin CustomSupports
2020-10-27 18:10:19,824 - DEBUG - [MainThread] UM.PluginRegistry.loadPlugins [432]: Loading all plugins took 0.6078441143035889 seconds
2020-10-27 18:10:19,847 - INFO - [MainThread] UM.VersionUpgradeManager.upgrade [163]: Looking for old configuration files to upgrade.
2020-10-27 18:10:19,920 - INFO - [MainThread] UM.VersionUpgradeManager.upgrade [175]: Checking and performing updates took 0.07463383674621582
2020-10-27 18:10:20,860 - DEBUG - [MainThread] UM.Settings.ContainerRegistry.loadAllMetadata [361]: Loading metadata into container registry took 0.7646839618682861 seconds
2020-10-27 18:10:20,872 - DEBUG - [MainThread] cura.Scene.CuraSceneController.setActiveBuildPlate [142]: Select build plate: 0
2020-10-27 18:10:20,874 - INFO - [MainThread] cura.CuraApplication.run [782]: Initializing machine error checker
2020-10-27 18:10:20,877 - INFO - [MainThread] cura.CuraApplication.run [787]: Initializing machine manager
2020-10-27 18:10:20,897 - INFO - [MainThread] cura.CuraApplication.run [792]: Initializing container manager
2020-10-27 18:10:21,132 - WARNING - [MainThread] LocalContainerProvider.LocalContainerProvider._saveCachedDefinition [295]: The definition cache for definition fdmprinter failed to pickle.
2020-10-27 18:10:21,759 - WARNING - [MainThread] cura.OAuth2.AuthorizationHelpers.parseJWT [119]: Could not retrieve token data from auth server: {"errors": [{"id": "ba5f6756-4dce-477d-9aa3-f75cc4f7262a", "code": "tokenExpired", "http_status": "403", "title": "The given token has expired.", "meta": {"error_type": "AuthError", "url": "https://account.ultimaker.com/check-token"}}]}
2020-10-27 18:10:21,762 - DEBUG - [MainThread] cura.OAuth2.AuthorizationHelpers.getAccessTokenUsingRefreshToken [61]: Refreshing the access token.
2020-10-27 18:10:22,324 - DEBUG - [MainThread] cura.OAuth2.AuthorizationService._storeAuthData [250]: Attempting to store the auth data
2020-10-27 18:10:23,485 - WARNING - [MainThread] UM.Logger.warning [128]: Starting a new sync while previous sync was not completed
{'CloudOutputDeviceManager': 0}
2020-10-27 18:10:23,759 - DEBUG - [MainThread] UM.Scene.Camera._preferencesChanged [255]: Changing perspective mode to perspective.
2020-10-27 18:10:23,799 - DEBUG - [MainThread] UM.Qt.Bindings.Theme.load [158]: Loading theme file: /Applications/Ultimaker Cura.app/Contents/MacOS/resources/themes/cura-dark/theme.json
2020-10-27 18:10:23,802 - DEBUG - [MainThread] UM.Qt.Bindings.Theme.load [158]: Loading theme file: /Applications/Ultimaker Cura.app/Contents/MacOS/resources/themes/cura-light/theme.json
2020-10-27 18:10:23,908 - DEBUG - [MainThread] UM.Qt.Bindings.Theme.load [246]: Loaded theme /Applications/Ultimaker Cura.app/Contents/MacOS/resources/themes/cura-light
2020-10-27 18:10:23,912 - DEBUG - [MainThread] UM.Qt.Bindings.Theme.load [246]: Loaded theme /Applications/Ultimaker Cura.app/Contents/MacOS/resources/themes/cura-dark
2020-10-27 18:10:24,214 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin._createView [340]: Creating post processing plugin view.
2020-10-27 18:10:24,216 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [202]: Begin loading of script: ChangeAtZ
2020-10-27 18:10:24,217 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [202]: Begin loading of script: ColorMix
2020-10-27 18:10:24,218 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [202]: Begin loading of script: DisplayFilenameAndLayerOnLCD
2020-10-27 18:10:24,218 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [202]: Begin loading of script: DisplayProgressOnLCD
[... 195 lines cut ...]
2020-10-27 18:18:26,524 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [/Applications/Ultimaker Cura.app/Contents/MacOS/resources/shaders/selection.shader]...
2020-10-27 18:18:26,528 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [/Applications/Ultimaker Cura.app/Contents/MacOS/resources/shaders/select_face.shader]...
2020-10-27 18:18:26,533 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [/Applications/Ultimaker Cura.app/Contents/MacOS/resources/shaders/default.shader]...
2020-10-27 18:18:26,537 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [/Applications/Ultimaker Cura.app/Contents/MacOS/resources/shaders/composite.shader]...
2020-10-27 18:18:26,541 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [/Applications/Ultimaker Cura.app/Contents/MacOS/resources/shaders/overhang.shader]...
2020-10-27 18:18:26,546 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [/Applications/Ultimaker Cura.app/Contents/MacOS/resources/shaders/striped.shader]...
2020-10-27 18:18:26,551 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [/Applications/Ultimaker Cura.app/Contents/MacOS/resources/shaders/transparent_object.shader]...
2020-10-27 18:18:26,555 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [/Applications/Ultimaker Cura.app/Contents/MacOS/resources/shaders/striped.shader]...
2020-10-27 18:18:26,560 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [/Applications/Ultimaker Cura.app/Contents/MacOS/resources/shaders/xray.shader]...
2020-10-27 18:18:26,564 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [/Applications/Ultimaker Cura.app/Contents/MacOS/resources/shaders/xray_composite.shader]...
2020-10-27 18:18:26,569 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [/Applications/Ultimaker Cura.app/Contents/MacOS/resources/shaders/platform.shader]...
2020-10-27 18:18:26,575 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [/Applications/Ultimaker Cura.app/Contents/MacOS/resources/shaders/xray.shader]...
2020-10-27 18:18:26,774 - DEBUG - [MainThread] cura.Machines.Models.CustomQualityProfilesDropDownMenuModel._update [33]: Updating CustomQualityProfilesDropDownMenuModel.
[... cut ...]
2020-10-27 18:18:36,414 - DEBUG - [MainThread] UM.Qt.QtApplication.windowClosed [470]: Shutting down cura
2020-10-27 18:18:36,455 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [315]: Attempting to kill the engine process
2020-10-27 18:18:36,455 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [321]: Killing engine process
2020-10-27 18:18:36,456 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [324]: Engine process is killed. Received return code -15

**Additional information**
(Extra information relevant to the issue.)

nallath, 2020-10-28:
Could you share your configuration folder?

oralallen82, 2020-10-28:
Here you go. 
[4.8 copy.zip](https://github.com/Ultimaker/Cura/files/5451584/4.8.copy.zip)

nallath, 2020-10-28:
Ah, i've found the issue. You created a printer in the addprinter dialog that has a dot in the name. That breaks the loading of the files on the next run.

oralallen82, 2020-10-28:
Ah....I did not know that. Let me correct and try again.

oralallen82, 2020-10-28:
Confirmed. We can close this issue.

closed 2020-10-28 by oralallen82

nallath, 2020-10-28:
Yeah, well its a bug on our side, it just took a bit more info to figure out what it was. 

It should be fine if you first create a machine and then re-name it though.

Ghostkeeper, 2020-10-29:
Printers with dots in their names work fine for me. Also after restarting Cura.

Ghostkeeper, 2020-10-30:
We figured out that the problem occurred only when the printer name started with a dot.

We opted for the easy fix of preventing printer names from starting with a dot. The issue isn't prevalent enough to fix the real issue, time-wise.

commit 31ee864f (2020-10-30): Disallow printer names that start with a dot

Fixes #8639.