Ultimaker/Cura #8641: 4.8.0 Z-Seam set to Sharpest Corner / Smart Hiding - layers do not all start at same location

msteele999, 2020-10-28:
**Application version**
4.8.0

**Platform**
Windows 7 Home 64 Bit / NVIDIA GTX970

**Printer**
Sovol SV-01

**Reproduction steps**
1. Load a model that has all layers exactly the same path
2. Set Z Seam Alignment = Sharpest Corner
3. Set Seam Corner Preference = Smart Hiding
4. Slice
5 Zoom into sharpest corner of model

**Screenshot(s)**
[image]

**Actual results**
When printing, patterns emerge on the print (see screenshot)
[image]

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**

[MacBook_Air_Stand.zip](https://github.com/Ultimaker/Cura/files/5448989/MacBook_Air_Stand.zip)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

Ghostkeeper, 2020-10-29:
What you're seeing in the screenshot is the result of wall overlap compensation. In this sharp corner the wall overlaps with itself for quite a while. To prevent overextrusion it's then reducing the flow on one of the sides.
[image]
This reduction of flow probably doesn't make a whole lot of difference in the final result though. The material flow out the nozzle is unable to change that quickly. It should get a better accuracy for edges pointing a bit upwards or downwards though, or if there are many such corners in a small space.

Unlike your example, for me the corner is completely consistent save for one single layer halfway:
[image]
In that layer it seems to have decided to compensate the other side of the corner.

Smart Hiding chooses to hide the seam with preference, but may also choose to expose the seam if there is a much sharper outside corner available. I think as a human I would choose the 90 degree inner corner here, but it seems that it always chooses the sharper outside corners here, so the decision is not as good as an expert, but I think that's acceptable. On my side it is stable though. Weird that there is such a difference between my result and yours, since [there hasn't been any commit on the 4.8 branch since the beta](https://github.com/Ultimaker/CuraEngine/commits/4.8) as of this testing.

msteele999, 2020-10-29:
Thank you for the explanation and the images - I would like to get mine to
look like yours - I wonder if there is a setting that causes the slicer to
do the overlaps all in one direction. Perhaps something innocuous that does
not logically stand out.

Mark

f1vefour, 2020-10-29:
I'm getting the exact same as msteele999 as you can see on the front right corner.

Ignore the layer artifact, the filament I'm using has foreign material in it that drags across the layers  #occasionally.

[image]

msteele999, 2020-10-29:
there it is! I have that on multiple prints.

Ghostkeeper, 2020-11-02:
@f1vefour Do you have a project file of that?

The calibration cube has 4 equal corners, so if the seam placement is just "Sharpest Corner" then it'll basically pick an arbitrary corner. Only by picking "User Specified" can you place the seam in a specific corner that guarantees it's the same corner.

msteele999, 2020-11-13:
Good morning fine folks at Cura - any ideas or updates on this issue?

msteele999, 2020-11-13:
Interestingly enough - in 4.8.0 release I know longer see this in the slicer - but I still see it in the print:

[image]

[image]

f1vefour, 2020-11-13:
I went back to an older version of branded Cura for now and have been using SuperSlicer as a stopgap for some things.

When this gets fixed I can happily go back to the latest version of Cura but as it is now I just can't use it.

msteele999, 2020-11-13:
> I went back to an older version of branded Cura for now and have been using SuperSlicer as a stopgap for some things.
> 
> When this gets fixed I can happily go back to the latest version of Cura but as it is now I just can't use it.
> 
> Besides this issue I get weird stops on corners which cause them to round off which you can also see a bit in my above image.

I've been doing the same thing. Unfortunately Cura 4.6.1 has a weird issue where it does not recognize the material type 'PLA+' - all my profiles disappear when I select that material type so I've had to name them differently.

I've also been using IdeaMaker - but would really prefer to stick with one slicer

Ghostkeeper, 2020-11-17:
Well, we haven't yet been able to reproduce this issue though. The original project file of the issue report didn't reproduce the issue any more in 4.8 stable, at least not on my side, and f1vefour hasn't provided reproduction steps yet. I've tried this with some of my own models as well where the shape of every layer is exactly the same, and it seems to always put the seam in the sharpest corner as far as I can tell, and the wall overlap compensation in the same place too.

Ghostkeeper, 2020-11-17:
> Interestingly enough - in 4.8.0 release I know longer see this in the slicer - but I still see it in the print:

Keep in mind that the banding you see there could just be because your infill is different in each layer. Infill provides resistance to warping. I don't really see anything wrong with the seam of that print, except that you might want to increase the jerk a little bit to make the corner a bit sharper.

f1vefour, 2020-12-14:
> Well, we haven't yet been able to reproduce this issue though. The original project file of the issue report didn't reproduce the issue any more in 4.8 stable, at least not on my side, and f1vefour [...]

I’m sorry that I didn’t provide more information at the time, I was not doing anything out of the ordinary or using a lot of settings. Basically it was the stock 0.2 Cura profile on a 20mm calibration cube. I don’t recall my exact seam settings but I believe it was sharpest corner and hide.

yngndrw, 2020-12-15:
@Ghostkeeper: I don't know if this is the same issue, something different or related to the [other 4.8 travel issue](https://github.com/Ultimaker/Cura/issues/8896), but I've noticed that my layers are starting on a straight line near a corner when set to start at the sharpest corner.

I've attached a file which has this issue - Please look at layer 1, the travel to the back of the the large square:
[image]
[CFFFP_Camera Mount.zip](https://github.com/Ultimaker/Cura/files/5696334/CFFFP_Camera.Mount.zip)

Ghostkeeper, 2020-12-18:
@yngndrw Your seam is set to the sharpest corner, but also to "hide seam". As a result, it's hiding the seam in the sharpest inner corner. It's not a straight line there, but there is an inner corner:

[image]

Setting the Seam Corner Preference to Smart Hiding will fix this issue and still behave similarly for other models as well (which do have strong inner corners).

yngndrw, 2020-12-18:
Thank you @Ghostkeeper, that makes perfect sense. Sorry for posting it in this thread, I thought it may have been a scenario for the same issue.