Ultimaker/Cura #8642: Cura 4.7.0 error message after installing 4.8b on the side

Thisismydigitalself, 2020-10-28:
**Application version**
4.7.0 / 4.8b

**Platform**
Win10/64/NV

**Printer**
Custom FDM

**Reproduction steps**
installed 4.8b without touching any other installed versions. now whenever i launch 4.7 i get the attached error

**Screenshot(s)**
[image]

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
4.7.0 should run without any errors. 

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save Project. For big files you may need to use WeTransfer or similar file sharing sites. G-code files are not project files!)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

Thisismydigitalself, 2020-10-28:
Uninstalling 4.8b and restarting windows eliminated this error message.
This is the first time i installed a newer version side by side to previous versions and got this error.

Ghostkeeper, 2020-10-29:
Error 0xc0000142 is Windows saying that the application is using a system DLL (in `C:\Windows`) that is [not present or not accessible](https://www.getdroidtips.com/fix-error-code-0xc0000142/). This can be caused by updates to Windows or to drivers. Cura uses graphics drivers, so it's possible that this happens during a run of Windows if the graphics drivers recently updated. Otherwise it can also happen with corrupted hard drives or if something went wrong during an update of Windows or a driver update. In any case, the most common advice I find on the internet is to [reboot Windows](https://www.techworm.net/2019/08/application-error-0xc0000142-windows-10.html).

It doesn't seem like anything that Cura can really do anything about. Cura requires the Visual C++ redistributable as well as working graphics drivers and a working Windows API. If any of those is missing or inaccessible, Cura is not expected to work.

closed 2020-10-29 by Ghostkeeper