Ultimaker/Cura #8643: [4.8 beta] Activate button isn't disabled for active machine profile

printingotb, 2020-10-28:
**Application version**
4.8 beta

**Platform**
 Windows 7, NVIDIA GeForce GT 730. (Using dual monitors).

**Printer**
Custom FFF or Creality cr10

**Reproduction steps**
Activate a printer profile

**Screenshot(s)**
Machine profiles menu:
[image]

The settings profile menu greys out the button:
[image]

**Actual results**
Button stays active

**Expected results**
Button should be disabled for the active machine.

**Project file**
[button.zip](https://github.com/Ultimaker/Cura/files/5452333/button.zip)

nallath, 2020-10-28:
Whoops. This has been blatantly wrong for over 4 years...

closed 2020-10-28 by nallath

fieldOfView, 2020-10-28:
... nervously checks git blame ...

Wasn't me!

printingotb, 2020-10-28:
While I'm at it, I found some inconsistencies that may or may not be intended :wink:

Add machine - it activates but doesn't automatically get selected

Create settings profile from overrides - activates and selects profile
Duplicate profile - doesn't activate or select

Create material - activates and selects
Duplicate material - activates and selects

Active machine can be removed and the next machine in the list activates. (Until now at least).
Active settings profile can't be removed
Active material can't be removed

nallath, 2020-10-29:
> Active machine can be removed and the next machine in the list activates. (Until now at least).
> Active settings profile can't be removed
> Active material can't be removed

This is intentional.

Ghostkeeper, 2020-10-29:
The inconsistency is necessary because while it's pretty clear what happens when you delete the active printer, a material or custom quality profile may be active on any number of printers, even ones that are not active right now. So deleting your custom material would also need to change the material on different printers. You wouldn't notice that this happens.

commit cc29ffa4 (2020-10-28): Disable the activate button if the machine is already active

Fixes #8643