Ultimaker/Cura #8653: [4.8] Discard current changes button works but feedback isn't provided

2357532, 2020-10-29:
**Application version**
4.8 Beta

**Platform**
W10

**Printer**
Generic

**Reproduction steps**
1. Apply "Discard current changes" button on Preferences/Profiles dialog

**Actual results**
Current changes to profile are discarded, however the Global Settings and Extruder tabs in the Preferences/Profiles dialog are not updated to reflect this.

**Expected results**
Current changes to profile are discarded, and the Global Settings and Extruder tabs in the Preferences/Profiles dialog are updated to reflect this.

**Log file**
2020-10-29 18:10:31,816 - DEBUG - [MainThread] cura.Machines.Models.QualitySettingsModel._update [76]: Updating QualitySettingsModel.
2020-10-29 18:10:31,819 - DEBUG - [MainThread] cura.Machines.Models.QualitySettingsModel._update [76]: Updating QualitySettingsModel.
2020-10-29 18:10:31,862 - DEBUG - [MainThread] cura.Machines.Models.QualitySettingsModel._update [76]: Updating QualitySettingsModel.
2020-10-29 18:10:33,626 - DEBUG - [MainThread] cura.Machines.Models.QualitySettingsModel._update [76]: Updating QualitySettingsModel.
2020-10-29 18:10:33,629 - DEBUG - [MainThread] cura.Machines.Models.QualitySettingsModel._update [76]: Updating QualitySettingsModel.
2020-10-29 18:10:33,633 - DEBUG - [MainThread] cura.Machines.Models.QualitySettingsModel._update [76]: Updating QualitySettingsModel.
2020-10-29 18:10:45,813 - DEBUG - [MainThread] cura.Machines.MachineErrorChecker._rescheduleCheck [155]: New error check scheduled.
2020-10-29 18:10:45,817 - INFO - [MainThread] cura.Machines.MachineErrorChecker._setResult [214]: Error check finished, result = False, time = 0.0s
2020-10-29 18:10:46,217 - INFO - [MainThread] CuraEngineBackend.CuraEngineBackend.slice [246]: Starting to slice...
2020-10-29 18:10:46,218 - WARNING - [MainThread] CuraEngineBackend.CuraEngineBackend.slice [250]: Slice unnecessary, nothing has changed that needs reslicing.

**Additional information**

konskarm, 2020-10-30:
I am unable to reproduce this issue. After I discard the changes, the values of the settings are properly set back to the default values for that profile. If close and reopen the Print Settings panel, the settings are still not updated to the proper default values of the profile you are switching to?

Ghostkeeper, 2020-11-02:
I think he's talking about this screen:
[image]

However I'm still unable to reproduce the issue. These are the steps that I took.
* Created a custom profile with adjusted infill density.
* Changed layer height to 0.12mm (up from 0.1mm).
* Opened up the preferences to the Profiles page, and verified that indeed the change to 0.12 shows up under the "Current Header".
* Closed the preferences screen again.
* Under the profile drop-down, clicked on "Discard current changes".
* Opened up the profile manager again, and verified that it indeed shows nothing any more under the "Current" header for layer height.

Could you maybe expand your reproduction steps to the full steps to what you did?

closed 2020-11-16 by no-response[bot]