Ultimaker/Cura #8654: Cura 4.7.1 with Creality CR-10 cannot print multiple items though there is no interference

mal98604, 2020-10-29:
**Application version**
4.7.1, 4.8

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

Win10

**Printer**
(Which printer was selected in Cura?)
Creality CR-10

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

1. Loaded model. Placed on front edge of plate
2. Loaded second model, placed on rear edge of build plate.
3. there is about 25 mm between the shadows of the 2 parts. plenty of clearance
4. Set print sequence to "one at a time"
5. Error, "Unable to slice"
6. Tried swapping model positions to see if it was a matter of print sequence causing collision. Same error

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 
[image]
[image]

**Actual results**
(What happens after the above steps have been followed.)

Error of "Unable to slice"

**Expected results**
(What should happen after the above steps have been followed.)

Should slice and let me examine layers.

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save Project. For big files you may need to use WeTransfer or similar file sharing sites. G-code files are not project files!)
[Cura log and 3mf for github.zip](https://github.com/Ultimaker/Cura/files/5461623/Cura.log.and.3mf.for.github.zip)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

See above zip file. attached. 

**Additional information**
(Extra information relevant to the issue.)

Please help. I have this giant printer and hate printing one piece at a time, when i could really fit 3 on the bed. Thanks!

fieldOfView, 2020-10-29:
It is a bit hard to tell from the top down view, but it looks like the models may be higher than the "gantry height" allows.

mal98604, 2020-10-29:
Hello!

I wouldn't think this matters because my print sequence would keep the gantry clear of the model. Please forgive me if I misunderstood.

Awesome response time!

Mark

GregValiant, 2020-10-30:
It does matter because the parts are 26.4mm tall and the Gantry height is defined as 25mm.  Cura sees a crash (between a finished part and the X beam) for the bottom 1.4mm of the second part.  Changing the gantry height to 27 allows the parts to slice.
With your print head at Z=0 carefully measure the distance from the X beam to the bed.  Like your CR10, my Ender gantry is defined as 25 but in reality it's 26.5.  A variable here is the nozzle length.  A longer nozzle would increase the Gantry height and the Initial Layer Height would increase the Gantry height by that amount.

Ghostkeeper, 2020-11-02:
Although it is possible that it may still print fine due to the exact shape of the gantry (e.g. having a gantry with just an X beam), Cura models the shape of the gantry as just a flat plane that covers the entire build plate. It doesn't know about the actual shape of the gantry.

Modelling the shape of the gantry goes into kinematics which is a) very complex and b) computationally expensive, especially in Python. We've considered this before, but decided that is out of scope. So just a flat plane it'll stay.

So this is then still expected behaviour. With a gantry height of 25mm, when printing one at a time, a model of 26.5mm is partially outside of the build volume and should be marked as such. There is one related issue to this still open: https://github.com/Ultimaker/Cura/issues/6627, which is still considered in scope but has no priority for us now.

closed 2020-11-02 by Ghostkeeper