Ultimaker/Cura #8658: Cura Menus Blank 4.7 & 4.6

playasyougo, 2020-10-30:
**Application version**
4.6 & latest 4.7 non beta

**Platform**
Windows 7 64bit RX550 GPU with latest drivers
Also updated Java

**Printer**
Creality Ender 5

**Reproduction steps**
1. Open any menu and it's blank

**Screenshot(s)**

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save Project. For big files you may need to use WeTransfer or similar file sharing sites. G-code files are not project files!)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues
[DxDiag.txt](https://github.com/Ultimaker/Cura/files/5465245/DxDiag.txt)
[cura log.txt](https://github.com/Ultimaker/Cura/files/5465246/cura.log.txt)
[image]

 to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

nallath, 2020-10-30:
Does it work after you resize the window?

playasyougo, 2020-11-02:
> 
> 
> Does it work after you resize the window?

If I resize there is sometimes a small amount of text that appears but not enough to be functional.

Ghostkeeper, 2020-11-02:
Duplicate of https://github.com/Ultimaker/Cura/issues/8216.

closed 2020-11-02 by Ghostkeeper

playasyougo, 2020-11-03:
> 
> 
> Duplicate of #8216.

Still not fixed it for me

Ghostkeeper, 2020-11-04:
Sure, but it helps to consolidate the discussions into one thread, to make it easier for people to share workarounds if they discover them, and to track the status of the issue better.