Ultimaker/Cura #8660: [4.8 beta] Add option to "Keep orientation" when arranging, duplicating and opening models

printingotb, 2020-10-30:
**Is your feature request related to a problem? Please describe.**
The first instance of a model that I drag into Cura is imported in the correct orientation. But every time that I add or multiply the same model, some of the copies are placed at a 180° angle (front becomes back).
(Multiplying grouped objects does not seem to change the orientation of the new group)

This messes up the layer seam placement in my current project as it needs to be on a specific edge of the model. So now I have to rotate and place all objects manually anyway.
Using the arrange all models function rotates some of them again.

**Describe the solution you'd like**
Add a check-box to the "Multiply Selected Model" dialogue under "Number of copies" : "Keep orientation of multiplied objects".
Add a dialogue with a check-box when clicking "arrange all models" to "Keep orientation of arranged models".
(There could also be other check-boxes there for features I or someone else may request later...)

**Describe alternatives you've considered**
Add a check-box in Preferences/"Opening and saving files" section to "Keep orientation when multiplying models" (this would include opened or drag and drop objects).
Also add a check-box in Preferences/"Opening and saving files" section to "Keep orientation when arranging objects."

**Affected users and/or printers**
Most users, whenever you need to print multiples of an object where the seam needs to be in a specific part of the model. Or other reasons like when a part cooler only cools from one side.

Ghostkeeper, 2020-11-02:
This was added to our backlog. We're adding a checkbox at least to the preferences screen but maybe also to the "multiply objects" dialogue.

Zogar89, 2020-11-11:
Now in the 4.8 final release there is no option to sort and keep rotation.

konskarm, 2020-11-12:
@Zogar89 that is correct. We have it in our backlog to be picked up sometime in the future.

Tetelot, 2021-04-21:
4.9 version and still rotating the objects.

bordermultimedia, 2021-04-27:
I would also like to upvote this, since it messe up things with multiple objects and seams defined to a specific location.

Tetelot, 2021-06-30:
4.10 version, still rotates the objects when multiplying or adding objects to the bed

At least it won't rotate the "previously rotated" objects. But I print figurines, so I always try to align the Z seam in the back of the pieces or at least where it will be easier to sand. 

Sorry by my bad English.

Ghostkeeper, 2021-07-07:
Yup, still on our backlog. Quite high though, in 55th place out of 339. I expect we'll be able to implement a feature for this in a couple of months unless the planning changes considerably.

topherbuckley, 2021-11-24:
May I ask why they are changed in the first place? Is this random or for some reason I'm not seeing? It seems to me it would take MORE work to rotate or otherwise transform the objects rather than just moving them via a simple translate transform, so I'd suspect there has to be some reason for this apparently random scattering of objects.

Ghostkeeper, 2021-12-03:
We are using a [library](https://github.com/Ultimaker/libnest2d) that not only finds the best position but also the best orientation for all objects. Indeed, one of the benefits is that it will also rotate objects for you, so that they fit more nicely. For most purposes, the orientation of the model around the Z axis doesn't matter much.

[image]

Zogar89, 2021-12-03:
I think z rotation is a bad idea.
The best I think is have the option to lock z rotation or not on
arragement.

topherbuckley, 2021-12-09:
> We are using a [library](https://github.com/Ultimaker/libnest2d) that not only finds the best position but also the best orientation for all objects. Indeed, one of the benefits is that it will also rotate objects for you, so that they fit more nicely. For most purposes, the orientation of the model around the Z axis doesn't matter much.
> 
> [image]
> 
> For some prints it does matter. For one it's annoying to have to move the seam position for each model, or rotate the skin pattern along for prints with highly optimised bridging. And for some prints the rotation was optimised for the directions in which the axes can accelerate fastest.

Thanks for the info. 

I'll add another case where it seems to matter; anything with supports. Support orientation makes a big difference when using zig-zag or the like. Also the spacing between models, or absolute position on the build plate perhaps? also seems to affect where the supports are generated, so having to re-calibrate this for every copy is quite the pain. Being able to copy then offset to a given position would eliminate this.

Tetelot, 2021-12-09:
Sorry by my bad english:

I print a lot of miniatures and monsters, around 120 per month.
It's a pain in the rear to use the new versions (4.8 beta, 4.8, 4.9.... and so on) with that automatic rotation.

So I still using the 4.7
I know it's a free program and "there are another priorities".
But come on, I tried to use the 4.12.

-If you try to multiply anything and you have at least one model outside the printbed, all the remain objects are put outside, even if you have nothing in the printbed.
-If you scale something and it's in the air, "arrange all models" won't put it in the floor, it'll still be in the air.

Tetelot, 2022-01-16:
4.13.0

Still the same problem.

[https://www.youtube.com/watch?v=QtsX35Q0BcI](url)

[https://www.youtube.com/watch?v=glpZEUiwzVE](url)

ysu2, 2022-04-24:
Guys, this is something that would be really-really highly appreciated as an option.    
I think one simple option; 
"Disable rearranging after multiplication" would be enough to help us out.

I mean look at this: even if I multiply already ordered pieces, the result is a mess. (probably in part because not all selected objects seem to be multiplied)  I'd prefer them to be multiplied w/o moving, or maybe a minimum offset to move them outside of the other object, but rotating and moving them into random positions is the worst option.  Lots of work to get them lined up.
[image]

smartavionics, 2022-04-25:
See https://github.com/smartavionics/Cura/releases for a Cura variant that provides settings to specify the distance between models and to disable rotations on arrangement. It's still a bit quirky but that's due to the arranging algorithm.

Tetelot, 2022-08-17:
> Yup, still on our backlog. Quite high though, in 55th place out of 339. I expect we'll be able to implement a feature for this in a couple of months unless the planning changes considerably.

Question: How is it going?

5.1.0 and still rotating without asking.

Nercury, 2023-04-23:
Three years later, this feature remains absent. Part orientation influences print quality due to varying ringing on different axes. Arranging parts in a neat grid enables easy counting and job scaling by adding or removing rows or columns. Auto-rotating is not a good default.

ungutknut, 2023-05-12:
+1 for this! Lately I spend way too much time canceling the rotation of print jobs with hundreds of parts.

printingotb, 2023-05-12:
> +1 for this! Lately I spend way too much time canceling the rotation of print jobs with hundreds of parts.

A trick I use is to place a number of parts in the correct orientation and create a group. Then you can multiply the group.

ferdbrowne, 2023-08-02:
same here, this auto rotate is complete nonsense

eGit, 2023-12-09:
5.4 still being a pain in the...

MariMakes, 2024-01-22:
Quick update from our side 👋
We are cleaning up our open issues, so we can focus on the most requested and needed features and bug fixes from our community and from UltiMaker, I stumbled upon your issue regarding model orientation when arranging and duplicating.

[image]

This is resolved in the 5.5 release with the grid arrange option, you can download the latest Cura version with the new feature here: [https://github.com/Ultimaker/Cura/releases/tag/5.6.0](https://github.com/Ultimaker/Cura/releases/tag/5.6.0)

I'll close this issue since it's resolved.
Thanks again, and please let us know if you run into any other issues 💪

closed 2024-01-22 by MariMakes