Ultimaker/Cura #8663: Orientation of FLSUN QQ-S Axes

LuizPonzini, 2020-10-30:
The orientation of the axes X and Y (on the table) are rotated 120º in relation of the FLSUN QQ-S Printer spectro.
Thank you.

Ghostkeeper, 2020-11-03:
In which direction is it rotated?

Are you able to fix the [mesh](https://github.com/Ultimaker/Cura/blob/master/resources/meshes/FLSUN-QQ-S.stl) for this printer yourself?

LuizPonzini, 2020-11-03:
Hi, thanks for your quick response.
Please see the correct orientation in the attached image.
I reduced the model for easier understanding, but the current Z position and scale are correct.
In fact, the angle difference is 135 ° (counterclockwise), not 120 ° as I said before.
Please contact me again if you want more details.
Thanks.
[image]

Ghostkeeper, 2020-11-04:
Thanks! I've changed it for version 4.9.

closed 2020-11-04 by Ghostkeeper

pmos69, 2021-01-14:
> Thanks! I've changed it for version 4.9.

There's an additional sub-issue related to this: the Printhead settings.
Those are still using the wrong orientation for the x/y axis.

Ghostkeeper, 2021-01-25:
> The Printhead settings.

If the print head shape needs to be rotated, I can't really help you there. I don't have that printer so I can't measure anything. I could blindly rotate the current rectangle around its coordinate origin but chances are that I'll rotate it the wrong way around or that it's not a rectangle and I'd have no way to know.

> The views selection buttons on the main windows (ex: "Top View")

These are aligned to the g-code coordinates, not necessarily to the front of the printer. Cura doesn't know what the front of the printer is (it just displays that 3D model) so there is no way to really change this to align to an arbitrary direction. We can't begin to try to detect where the LCD display is in a 3D model or anything.

> Also, it's still off by 25º.

You are then disagreeing with @LuizPonzini. Could you come to an agreement with them?

pmos69, 2021-02-22:
> > The Printhead settings.
> 
> If the print head shape needs to be rotated, I can't really help you there. I don't have that printer so I can't measure anything. I could blindly rotate the current rectangle around its coordinate origin but chances are that I'll rotate it the wrong way around or that it's not a rectangle and I'd have no way to know.
> 
> > The views selection buttons on the main windows (ex: "Top View")
> 
> These are aligned to the g-code coordinates, not necessarily to the front of the printer. Cura doesn't know what the front of the printer is (it just displays that 3D model) so there is no way to really change this to align to an arbitrary direction. We can't begin to try to detect where the LCD display is in a 3D model or anything.
> 
> > Also, it's still off by 25º.
> 
> You are then disagreeing with @LuizPonzini. Could you come to an agreement with them? And are you sure you've tested with version 4.9 or master? It's not been released yet.

Hi @Ghostkeeper,

Sorry for the confusion. I don't think I was clear.
I tested with Arachne engine alpha (99.0.0 I think).

Btw, I perfecly understand that you can't determine what is the "front" side of the printer

Here's the problem: With the rotation I see in Arachne the "problem" is actually worse than before.

This is an example stl in 4.8.0 in top view:
[image]

Sure, the STL is not correctly orientated because the "DOWN" side is not facing the "front" of the printer, but at least it's correctly facing one of the printer sides. And that is how it prints: That side of the printed model is parallel to the printer side because the x axis runs paralel to the printer side panel. (even if it's the wrong side).

It's anoying that it's the wrong side? Yes, but I guess it can be fixed by shifting the connections of the 3 stepper motors to get the orientation right.
I don't even know if it's a general issue or just poor quality control by the manufacturer.

In Arachne, the same model in top view looks like this:
[image]

It's still not facing the right printer side, but the real problem is, it's now not even paralel to the printer side in the representation.

PS: If you want to know, in my printer the "front" side is this one, but as I said, that a minor issue:
[image]

Thanks.

commit 9ffc6bc7 (2020-11-04): Rotate FLSUN-QQ mesh by 135 degrees around negative Z

Also convert it to 3MF, to save some file size.
According to reports in #8663 this is the true orientation of the printer w.r.t. the coordinate system in its firmware.

Fixes #8663.