Ultimaker/Cura #8664: Cura 4.8Beta - SideBar plugin installed but not working

Thisismydigitalself, 2020-10-30:
**Application version**
4.8 Beta

**Platform**
Win10/64/NV

**Printer**
Custom 

**Reproduction steps**
installed 4.8 side by side to 4.7.0, expecting to have this good old sidebar which is installed to no avail.

**Screenshot(s)**
 
[image]

**Actual results**
Expecting to have the sidebar working as i just can't work with 4.x default settings menu - i just can't stand having the layer slider on the right of this setting menu which makes absolutely no sense to me whatsoever.

**Expected results**
I sure hope  4.8 released version will support this plugin.

nallath, 2020-10-30:
Please report this with the github of the sidebar plugin: https://github.com/fieldOfView/Cura-SidebarGUIPlugin/

closed 2020-10-30 by nallath

fieldOfView, 2020-10-30:
A test version of SidebarGUIPlugin that is compatible with Cura 4.8 is avilable here:
https://community.ultimaker.com/topic/26014-sidebar-gui-for-cura-4/page/4/?tab=comments#comment-275194

For details, see https://github.com/fieldOfView/Cura-SidebarGUIPlugin/issues/17

The next version of SidebarGUIPlugin is awaiting review for release in the Marketplace