Ultimaker/Cura #8665: Cura 4.8 Beta - Startup Optimiser not working.

Thisismydigitalself, 2020-10-30:
**Application version**
4.8 Beta

**Platform**
WIn10/64/NV

**Printer**
Custom

**Reproduction steps**
**Startup Optimiser** installed but does not work as expected, meaning despite disabling affected profiles, each time I launch 4.8 this message appears. This plug works perfectly with 4.7.x

**Screenshot(s)**
 
[image]
[image]

nallath, 2020-11-02:
Please report issues with bugs in the correct repository. In this case: https://github.com/fieldOfView/Cura-StartOptimiser/

closed 2020-11-02 by nallath

fieldOfView, 2020-11-02:
Did you try the "Disable affected profiles" button? That is provided by the StartOptimiser plugin.
As far as I can see, the plugin is working as expected.

Thisismydigitalself, 2020-11-03:
> 
> 
> Did you try the "Disable affected profiles" button? That is provided by the StartOptimiser plugin.
> As far as I can see, the plugin is working as expected.

Of course. this plug works as expected in 4.7.0. not so with 4.8b. 4.8b keeps showing this list of issues each time I launch it.

Thisismydigitalself, 2020-11-03:
> 
> 
> Please report issues with bugs in the correct repository. In this case: https://github.com/fieldOfView/Cura-StartOptimiser/
If you ask me this is a Cura issue cause these errors should not be there in the first place. Cura can't be fixed by thousands of plugins which we the users should guess what causes and what fixes these issues. we are normal users which do not always now what goes on under the hood. I believe i saw someone commenting that these errors should be a thing of the past starting from 4.7x or 4.8.  Plugins are great for temporary patches but it is very important to not get used to fix Cura core functionalities with adding features. getting read of error messages using a plugin is great, but not a solution.

fieldOfView, 2020-11-03:
> If you ask me this is a Cura issue cause these errors should not be there in the first place. 

Then rephrase your problem. Making sure a plugin works is the responsibility of the plugin author. Making sure Cura doesn't mangle your configuration is the responsibility of Cura devs.

> Plugins are great for temporary patches 

I consider very few (if any) of my plugins temporary patches. I like to think of them as adding missing functionality.

Thisismydigitalself, 2020-11-03:
> > Plugins are great for temporary patches
> 
> I consider very few (if any) of my plugins temporary patches. I like to think of them as adding missing functionality.

missing functionality - i accept your correction hoping you dig my general idea.
I love your plugins and your contribution - i just thought that me reported this issue to the cura team was and still is me doing the right thing.

nallath, 2020-11-03:
> missing functionality - i accept your correction hoping you dig my general idea.
> I love your plugins and your contribution - i just thought that me reported this issue to the cura team was and still [...]

As @fieldOfView already mentioned: statements like "Plugin X isn't working" isn't something that should be mentioned to the Cura developers (unless we are the authors of that plugin). But something like "Plugin X does something that should be a main feature of Cura" is something that should be mentioned  (since that's a feature request)