Ultimaker/Cura #8666: Can't import profiles on 4.7.1 (for linux)

SeaGull711, 2020-10-30:
**Application version**
4.7.1

**Platform**
Ubuntu Linux (20.10)

**Printer**
Creality Ender 3 V2

**Reproduction steps**
1. Preferences > Configure Cura > Profiles > Import Profile
2. Try to import Mark Gonsowski's Litophane Profile from https://www.gameon3d.com/downloads-links (HERE)

**Screenshot(s)**
[image]
[image]

**Actual results**
Profile isn't added/available/visible

**Expected results**
Profile would become available to select under "custom profiles"

**Project file**
N/A

**Log file**
>2020-10-30 15:43:11,382 - DEBUG - [MainThread] cura.Machines.Models.QualitySettingsModel._update [76]: Updating QualitySettingsModel.
>2020-10-30 15:43:11,389 - DEBUG - [MainThread] cura.Machines.Models.QualitySettingsModel._update [76]: Updating QualitySettingsModel.
>2020-10-30 15:43:11,514 - DEBUG - [MainThread] cura.Machines.Models.QualitySettingsModel._update [76]: Updating QualitySettingsModel.
>2020-10-30 15:43:17,757 - DEBUG - [MainThread] cura.Settings.CuraContainerRegistry.importProfile [186]: Attempting to import profile /home/me/Downloads/Gonsowski Liquid Metal Lithophane.curaprofile
>2020-10-30 15:43:17,851 - DEBUG - [MainThread] cura.Machines.Models.CustomQualityProfilesDropDownMenuModel._update [33]: Updating CustomQualityProfilesDropDownMenuModel.
>2020-10-30 15:43:17,870 - DEBUG - [MainThread] cura.Machines.Models.QualityManagementModel._update [306]: Updating QualityManagementModel.
>2020-10-30 15:43:17,884 - DEBUG - [MainThread] cura.Machines.Models.CustomQualityProfilesDropDownMenuModel._update [33]: Updating CustomQualityProfilesDropDownMenuModel.
>2020-10-30 15:43:17,895 - DEBUG - [MainThread] cura.Machines.Models.QualityManagementModel._update [306]: Updating QualityManagementModel.

**Additional information**
I've been reading through all the previous bug reports and I see some similar issues, but, unlike #8229, this is not an upgrade, it's a brand new install.  I have NOT setup any custom materials or nozzles or anything beyond configuring my printer (An Ender 3 V2). 

I did determine that the problem occurs if I unpack the .curaprofile file and edit the files inside so that the line "quality_type = draft" instead reads, "quality_type = standard", and repack it, it seems to import as it should, but, correct me if I'm wrong, I'm assuming that Cura is filling in any unspecified settings using the existing profile, either "draft" or "standard", and making this change will result in a combination of settings that were not intended to work together.  So... does the "draft" profile exist under a different name now?  Or, what can I do to get this working properly?

nallath, 2020-11-02:
That's because the profile you are importing is made for a profile called "draft", whereas your currently active printer doesn't have a profile named like that. 

So basically; you are trying to import a profile that isn't suited for the printer that you have active.

Ghostkeeper, 2020-11-03:
In Cura 4.8 we've clarified this error message better, so that it should be clear why the profile isn't currently visible

closed 2020-11-03 by Ghostkeeper