Ultimaker/Cura #8667: Infill Density

viper0078, 2020-10-31:
**Application version**
4.7.1

**Platform**
Win64

I'm not sure if this is a bug or not, so let me ask you a question.
If I change the value of InfillDensity, the density does not change.
Changing the InfillLineDistance will change the density.

It appears on the left side of the text box where you enter the value.
I don't understand the arrows and fx buttons either, as they seem to only return to the initial values.
Is it a specification that the value of InfillDensity(%) is not reflected even if it changes? Is it a bug?

fieldOfView, 2020-11-01:
The `Infill Line Distance` is normally calculated from the `Infill Density` (and the `Line Width` and the 'Infill Pattern'). However, once you put a "literal" value in the Infill Line Distance, this overrides the calculated value. I think there is a literal value in the profile you selected.

viper0078, 2020-11-02:
Thanks for the reply.
I've downloaded and used the profile for Snapmaker, which means it's customized.
I used the Default profile in PrinterSettings and was able to get the two parameters to work together.
Thank you.

closed 2020-11-02 by viper0078