Ultimaker/Cura #8669: Bridged Surface over Infill direction changes

Sophist-UK, 2020-10-31:
**Application version**
4.7.1

**Platform**
Win-64

**Printer**
Dagoma DiscoEasy 200

**Reproduction steps**
1, Surface Lines [45,90,135,0]
2. Cubic infill at 20%
3. Minimum Bridge Wall Length 2mm
4. Set Bridge Sparse Infill Maximum Density to 0% then 40%

**Screenshot(s)**
*Bridge Sparse Infill Maximum Density 0%*
[image]

*Bridge Sparse Infill Maximum Density 40%*
[image]

**Actual results**
Line direction changes when Bridging over infill is enabled.

**Expected results**
Line direction should stay the same when Bridging over infill is enabled.

**Project file**
[threadholder.zip](https://github.com/Ultimaker/Cura/files/5469077/threadholder.zip)

**Log file**
Nothing relevant in log file

Sophist-UK, 2020-10-31:
Note the direction with bridging is NOT one of the directions actually set. Here is the screen shot for 2 layers later:
[image]

Ghostkeeper, 2020-11-03:
It's changing the direction of these lines *because* it is bridging. Changing the direction of these lines is exactly what the bridging routine does. I'd say this is expected behaviour. What would you expect that bridging does?

Sophist-UK, 2020-11-03:
Yes - now that you prompt me to think about it, I can see that I would expect it to change direction in some or even most circumstances. 

If we consider a bridge in the accepted sense i.e. long and narrow, it would not make sense to try to lay down filament across the width of the bridge rather than the length.

In a more generalised case, I imagine that the bridging routine would be interested in finding the direction which: A) starts and stops on immediately underlying material i.e. existing in the immediately previous layer; and B) results in the shortest distances to bridge.

However, in the case where we are bridging over infill, or perhaps specific types of infill (e.g. cubic in my case) where there may not be any one direction that is significantly better than any other, should we change direction for no benefit or for trivial benefit (and possibly go against the users desire for cross-hatched or parallel layering) or should we **not** change direction in these circumstances?

NOTE: This was always a low priority issue - but I think this discussion makes it even lower priority.

Ghostkeeper, 2020-11-04:
Maybe.

Normally I would only really use this feature when printing at very low infill densities or when there is no infill at all. The normal skin printing works well and is also adjustable in many ways. Just not the fan speed and density, really. So without adjusting the orientation, there is very little reason left to use the feature I think.

Sophist-UK, 2020-11-04:
I was using this feature in order to slow down the print on the first top layer after infill. I have my printing tuned to go much faster than defaults, and without using the bridging functionality the first top surface layer droops into the infill - and then that provides a decent base for the 2nd top surface layer, so the net result is that I need an additional top surface layer. 

So the feature does work for me, and a different print direction is only a very minor thing.