Ultimaker/Cura #8670: "One at a time" setting enabled, but print all first layers at once

Sebbb, 2020-10-31:
Hello,

I like the "all at a time" feature as it heavily reduces movements and thus saves printing time, when objects are placed correctly.

However, as it prints multiple times the first layer, I have to constantly monitor if the first layers are printed correctly. Especially the first layers tend to not stick to the plate correctly. 

So, it would be nice to have a setting "Print all first layers in the beginning", and then continue printing the complete objects, one at a time.

printingotb, 2020-11-02:
I like this idea. 
An alternative would be if we could print a common skirt that goes around all parts to see if the bed level is correct.

konskarm, 2020-11-03:
Hi @Sebbb. We have considered it before as an option, but the engine would need a significant refactor to accommodate for that, so we deferred it. 
I am closing this as a duplicate of https://github.com/Ultimaker/Cura/issues/7440. Please add other comments in that issue.

closed 2020-11-03 by konskarm