Ultimaker/Cura #8671: Ultimaker Cura 4.4.1 on Ubuntu 20.04 looks unfinished. No progress immage. Print Progress bar writing over the top of Jog/Send Code boxes.

KevinParish01, 2020-10-31:
**Is your feature request related to a problem? Please describe.**
(A clear and concise description of what the problem is. Ex. I'm always frustrated when [...])

**Describe the solution you'd like**
(A clear and concise description of what you want to happen. If possible, describe why you think this is a good solution.)

**Describe alternatives you've considered**
(A clear and concise description of any alternative solutions or features you've considered. Again, if possible, think about why these alternatives are not working out.)

**Affected users and/or printers**
(Who do you think will benefit from this? Is everyone going to benefit from these changes? Or specific kinds of users?)

**Additional context**
(Add any other context or screenshots about the feature request here.)
[image]

fieldOfView, 2020-11-01:
I'm guessing (because you didn't fill out the template that's all I can do) that you are not using the AppImage? Only the AppImage is supported by Ultimaker.

KevinParish01, 2020-11-01:
Hi Aldo,
Thank you for pointing out my error.
Thanks for the prompt response.

I downloaded both some time ago and thought the installed version was the
same as the appimage so deleted the appimage.

Downloaded the appimage again.
Layout looks much better.

Why so much white space on the monitor screen?

Kind regards
Kevin.

Ghostkeeper, 2020-11-03:
Most likely because Ubuntu 20.04 would use Qt 5.12 while Cura is built against Qt 5.10. This sort of issue is the major reason why we distribute an AppImage with fixed versions of (almost) all dependencies, rather than having to support many different versions at the same time.

closed 2020-11-03 by Ghostkeeper