Ultimaker/Cura #8672: [4.8.0 beta] Top/Bottom Patterns non-connecting

colatuttorf, 2020-11-01:
**Application version**
4.7.x and 4.8.0 Beta

**Platform**
Mac OSX 10.13.6

**Printer**
Robo R2 (Custom FFF)

**Reproduction steps**
1. Slice Model
2. Click Preview
3. Look through layers, 2+ show either excessive movements with Top/Bottom Pattern: Lines (image 1, below) or disconnected sets of zig zags with Top/Bottom Pattern: Zig Zag (image 2, below)

**Screenshot(s)**
[image]
[image]

**Actual results**
Printer makes excessive extra travels and the disconnect between the top/bottom lines/zig zags is visible, sometimes shows through walls

**Expected results**
these should all be connected and create a continuous top/bottom line or zig zag pattern

**Project file**
paid models/products, can't post.  Models were exported from Onshape, if that helps.

Ghostkeeper, 2020-11-03:
You might have Top Surface Skin Layers that need to be printed separately. Without a project file, it's impossible to see. I can't reproduce the issue with the defaults for Custom FFF Printer in any case:
[image]

Maybe it reproduces with a simple model that you can share, like a pyramid?

colatuttorf, 2020-11-03:
You're right, it does go away when I set Top Surface Skin Layers to 0.

I can make a model that shows this and post a project, good idea.  I think I've figured it out though, once you said to check the surface skin layers.  I have thin bottoms on my models that print in 3 layers, so those volumes are all Top/Bottom layers (with a skin layer), and they're adjacent to taller objects which are trying to switch to normal Top/Bottom layers, and the Skin Layers setting is slicing the solid volume up as shown above.

I feel like this behavior shown isn't necessarily the intent though, so not sure I'd remove the 'bug' tag, but maybe there needs to also be a feature request generated for something like "Extend Skin Layers Through Adjacent" which makes any neighboring top/bottom lines also skin lines so they can be printed continuously.

colatuttorf, 2020-11-03:
Here is a project file that replicates the behavior:

[CFFFP_Cura_Tests-SurfaceSkinInteraction.3mf.zip](https://github.com/Ultimaker/Cura/files/5482019/CFFFP_Cura_Tests-SurfaceSkinInteraction.3mf.zip)

Ghostkeeper, 2020-11-05:
The top surface is split off from the normal top skin in order to allow printing with different settings. You can imagine that if you use a different line width for the top surface, there is no way in which we can make the lines connect. The same goes for the pattern or the line directions.
For the flow and speed settings it can theoretically be done, but you probably don't *want* to print them intermixed. That would require the flow rate out the nozzle to change back and forth halfway every line segment, which is not going to go over well due to the delay in the feeder instruction and the actual flow rate of material out of the nozzle opening.

If all settings for top skin are the same, we could make them join together. But of course there is no reason for a user to have top surface skin but then not use different settings for it, so I don't think that use case is worth us spending time on. For that reason I'll close this issue then. Sorry!

closed 2020-11-05 by Ghostkeeper