Ultimaker/Cura #8675: [4.8 beta] Settings guide blanked out words

printingotb, 2020-11-02:
**Application version**
4.8 beta

**Platform**
win7

**Printer**
any

**Reproduction steps**
1. Enable "show articles in settings tooltips"
2. Open Settings visibility in preferences, hover the cursor over settings.

**Screenshot(s)**
[image]

**Actual results**
words blanked out

**Expected results**
not blanked out

**Additional information**
Side-note: the text under the left picture of "Connect Top/bottom Polygons says "Naive concentric pattern", should it be "native"?

And by the way, is reporting these kind of bugs overkill?

Asterchades, 2020-11-02:
> And by the way, is reporting these kind of bugs overkill?

No, but reporting them here isn't appropriate. The Settings Guide isn't part of Cura - it's an extension.

You'll want to make a report on the repo responsible for the extension: https://github.com/Ghostkeeper/SettingsGuide

printingotb, 2020-11-02:
I see, how do I delete this post?

Asterchades, 2020-11-02:
Honestly? I'm not entirely sure that you can. But you can close the issue, which is mostly the same thing in the grand scheme of things. There should be a button down with the one you use to post a comment allowing you to do exactly that (either by itself or with a response).

printingotb, 2020-11-02:
Found it :sweat_smile: