Ultimaker/Cura #8680: [4.7.1] Weird extruder switch while printing with a raft

InsightStu, 2020-11-03:
**Application version**
Cura 4.7.1

**Platform**
Win 7 Professional 64bit | Xeon E3-1270v5

**Printer**
Custom FFF printer with 3 extruders, a mix of custom and generic materials.

**Reproduction steps**
The issue presents itself every time I print with a raft. 
While printing the first and second layer with a custom material on extruder 2, extruder 1 heats up, briefly positions itself close to the starting point for that layer, does something I can't seem to comprehend and then switches back.
This is an issue for me as the material in extruder 1 degrades with heat and the chamber empties and oozes, the whole goal of the process is to do material caracterization, so i need a very repeatable sample of fresh material. I get around this by purging but this is not very consistent for me due to limitations on my direct drive.
Right now this has only happened to me if I print with a raft, which is something I have to do for the same repeatability reasons.

**Screenshot(s)**
[image]
[image]
Apologies for my circle drawing skills, or lack thereof.. 😄 

Gcode looks like this, I will also attach the full file:
[image]
[image]

**Expected results**
Extruder 2 prints the raft, then extruder 1 switches when it really is its turn.

**Project file & gcode**
[Issue Report - Tensile Project.zip](https://github.com/Ultimaker/Cura/files/5480942/Issue.Report.Project.zip)
[Issue Report - GCODE.zip](https://github.com/Ultimaker/Cura/files/5480994/Test.13.-.no.fan.abs.top.zip)

**Log file**
I don't think its needed, but I can provide it if necessary!

**Additional information**
I checked for closed and open issues and I have found nothing similar, I hope this is not a duplicate.
Also, just wanted to say thanks to all the devs involved with Cura, I'm currently in the process of converting all the printers at work to Cura from another slicer, and I hope I can contribute even more to this project as time goes on in the form of some PRs and new printers. I'm currently working on the jsons to get away from the Custom FFF profile and streamline my machine lineup.

Ghostkeeper, 2020-11-05:
This is a very good, extensive bug report. Thank you.

I can reproduce your bug using your project file. (For other people: make sure you disable dropping models to build plate and making sure they don't collide in your preferences, before loading the project file.) I have little idea why it would print something there, but I can see two things that might be a hint:
* It's marked as a helper structure in the g-code.
* This only happens if the prime tower is enabled.

I'll bring this up with the other developers to determine priority.

StuSerious, 2020-11-05:
Thanks Ghostkeeper! 
In the meantime I have kinda found a workaround (as in just purge 100mm^3 of material on the tower and be done with it 😆) 
I can afford this now because I'm just printing simple TPU, but I also work with MUCH more expensive and MUCH more abrasive materials. But I guess I'll think about it when I cross that bridge.
For now, I wouldn't flag this as high priority since it seems to be an edge case and, while it is repeatable, I don't think it's affecting a lot of users out there!
In the weekend I'll try to poke around in the code and see what perfect storm is causing this, but I'm not confident I'll find anything. If I do, I'll let you all know!
Anyways, thanks for your time and I'll patiently wait to see if there is a solution!

Ghostkeeper, 2020-11-10:
Considering the rarity of this case and the unintended effect that it adds a minute or so to the printing time, and considering there are workarounds, we've decided to defer this issue. It's indeed a bug, but we don't consider it important enough to be working on now. We get many more issue reports than what we can fix, and this isn't a priority compared to the rest. We'll keep it open in case priorities change though.

StuSerious, 2020-11-11:
Roger that! 
I'll keep testing every version and I'll update this thread in case something ever changes.
In the meantime, thanks for the time and I'll keep y'all posted

StuSerious, 2020-12-22:
Update 1]
Finally got time to switch to 4.8, problem still presents itself. I started to look into the code a bit more closely, still nothing comes to mind tho..
I shall post new updates on new releases and/or if I figure out what's going on!

StuSerious, 2021-02-23:
@Ghostkeeper Hey there! Sorry for the ping!

I have another issue similar to this one, the triggering conditions seem to be the same as described above (tho I'm not 100% sure), so I want to ask you first if I have to open a separate Issue Report.

I'm getting this weird extrusion even when extruder 1 [T0 in my case] is _disabled_, take a look:
[image]
The material in gray is HIPS [Ext2 - T1], and the one in green is PC-ABS [Ext3 - T2]. There should not be any Ext1 in there, but still it lays down a whole patch of material, in addition to all the behavior described in this Issue with all the periodically heating up the extruder for no apparent reason..

I'm getting more and more confused since I really have no clue about what's causing this.. things seem to be breaking down more and more D:
What do you think? Could this be a separate Engine-related issue or should I post a more detailed report in this thread?

Oh and [here](https://we.tl/t-1iYmUCXSN0) you can find the .GCODE if you want to take a look for yourself!

Thanks for your time, and sorry again for the ping 😅

StuSerious, 2021-03-02:
@Ghostkeeper 
Pardon the ping again, @ing for visibility!

Ghostkeeper, 2021-03-03:
I'm purposefully a week behind on the Cura issue notifications to allow other people to react first, sorry. Just seeing your first ping now.

I think your new issue is already reported here: https://github.com/Ultimaker/Cura/issues/8837

StuSerious, 2021-03-09:
Oh sorry if I bothered you :/

I'll check that out and update there! Thanks a lot!

MariMakes, 2023-06-27:
Hey @StuSerious,

👋 A fresh set of eyes reporting for duty 💪 

Would you mind double-checking this behavior in the Cura 5.4 Beta?
We introduced changes related to the printing order of the prime tower, and heating up and cooling down behaviors. 
[You can read more about the details here. ](https://github.com/Ultimaker/CuraEngine/pull/1798)

Can you let us know if that resolves your issue? 
You can download the Beta here: https://github.com/Ultimaker/Cura/releases/tag/5.4.0-beta.1

If it does not resolve it, please let us know and I'm happy to take another look.

closed 2023-07-11 by github-actions[bot]