Ultimaker/Cura #8682: [4.8 beta] "Discard changes" causes print sequence go blank (settings updated to match the availablity of extruders)

printingotb, 2020-11-03:
**Application version**
4.8 beta

**Printer**
At least Custom fff, Anycubic kossel linear plus, wanhao duplicatori3.
Only with my old settings-profiles created in 4.3 (or possibly even older but imported and exported from 4.3)
The printer profiles are created fresh in 4.8

**Reproduction steps**
1. open attached project
2. change any setting and then hit discard current changes

**Screenshot(s)**
[image]
[image]
[image]

**Project file**
[Settings updated.zip](https://github.com/Ultimaker/Cura/files/5481762/Settings.updated.zip)

**Log file**
[stderr.log](https://github.com/Ultimaker/Cura/files/5481770/stderr.log)

**Additional information**
Apart from this, the profiles work as they should. If it's simply something wrong with my profiles, I can edit my files if I know what to change.
After uploading this, I changed quality type fast to normal in all the profile files, it made no difference.

nallath, 2020-11-03:
It seems to have been broken by the changes made in https://github.com/Ultimaker/Cura/commit/9266c753c30868ca3b6badda41da1b0f6d9b9857

I'll discuss this with the rest of the team!

printingotb, 2020-11-04:
Thanks :smiley:

jellespijker, 2020-11-10:
fixed for 4.8

closed 2020-11-10 by jellespijker