Ultimaker/Cura #8683: [4.8 beta] Connect Top/bottom polygons. Setting "top surface pattern"- or "bottom pattern inital layer" to concentric does not make the setting visible

printingotb, 2020-11-03:
**Application version**
4.8 beta

**Printer**
any

**Reproduction steps**
1. Set top/bottom pattern to lines
2. Set Bottom pattern initial layer and/or Top surface skin pattern to concentric

**Screenshot(s)**
[image]
[image]

**Actual results**
Connect Top/bottom polygons setting is hidden
(However, it is still active and working)

**Expected results**
Setting should be visible

**Additional information**

Ghostkeeper, 2020-11-05:
Yup, can confirm. It still works but it was hidden.

We can easily change the visibility condition there. I've fixed it in master (so 4.9).

You're on a roll recently. Thanks for all the bug reports!

closed 2020-11-05 by Ghostkeeper

printingotb, 2020-11-05:
> You're on a roll recently. Thanks for all the bug reports! 

You are welcome! Thanks for an amazing software :smiley:

commit a4b2dc2c (2020-11-05): Show Connect Top/Bottom Polygons when top surface pattern is concentric too

If there is a top surface, that is.

Fixes #8683.