Ultimaker/Cura #8690: infill

therealbirdman, 2020-11-04:
Unable to adjust infill percentage in version 4.8 beta

StuSerious, 2020-11-04:
Adjusting infill works fine for me, and this is very non descriptive, could something like #8667 be your issue?

nallath, 2020-11-04:
It seems that the issue template has not been filled in entirely. Please check our guide on https://github.com/Ultimaker/Cura/wiki/Reporting-issues for more information on what information we need and why we need it.

therealbirdman, 2020-11-04:
Not sure why it is grayed out?, load the same file in 4.7.1 or lower and able to adjust, it is at a 10% value need to set too 100%

nallath, 2020-11-05:
I am also not sure why it's greyed out, in order to be sure about it we will need the information that the template asks for. 

Without that information, I can only speculate. It's probably because you have changed the infill line distance. When you hover over the infill density you will also see this in the tooltip (it states "This setting is not used because all the settings that it influences are overridden")

closed 2020-11-19 by no-response[bot]