Ultimaker/Cura #8692: tabs in Cura

omega-regulus, 2020-11-04:
Hello, I was on windows and I become a Mac user recently and I have "recommandation" for you, your software is so cool but can you put different tabs like a web browser to switch between printers and stl files without make a new project and keep a sliced stl while slicing another for example.

Glodigit, 2020-11-04:
+1 on this. Really like Mattercontrol's use on this, especially since the multi build plate feature of 3.2 seems to have been removed.

Ghostkeeper, 2020-11-05:
Duplicate of https://github.com/Ultimaker/Cura/issues/7009.

Sorry, this was considered out of scope for the time being. It's an enormous task and regression hazard to implement something like this. For now, the workaround is to just open multiple instances of Cura.

closed 2020-11-05 by Ghostkeeper