Ultimaker/Cura #8694: 2nd extruder temperature setting is used for first few layers of first extruder

matthuszagh, 2020-11-04:
**Application version**
4.7.1

**Platform**
Linux 5.9.1 (on NixOS)

**Printer**
MendalMax3

**Reproduction steps**
The printer has dual extruders. The extrusion temperature for the first extruder is set to 260 (for a nylon filament) and the second extruder has its temperature set to 220 (for a PVA). Even if the second extruder is not used, but is active, the first ~2 layers of the first extruder print at 220 before rising to 260. With the identical setup by with the second extruder inactive, the first extruder only prints at 260.

Here are my temperature settings for the first extruder, 

[image]

and here they are for the 2nd extruder

[image]

Here's what I get as the first part of a print (just a simple rectangular box):
<details>
```
;FLAVOR:Marlin
;TIME:320
;Filament used: 0.167156m, 0m
;Layer height: 0.1
;MINX:115.2
;MINY:150.2
;MINZ:0.3
;MAXX:134.8
;MAXY:169.8
;MAXZ:1
;Generated with Cura_SteamEngine 4.7.1
T0
M82 ;absolute extrusion mode
G28
M104 S150
M190 S50.0
G90
G92 E0
M109 S220.0
G92 E0
G92 E0
G1 F2400 E-6.5
;LAYER_COUNT:8
;LAYER:0
M107
;MESH:20x20x1mm_first_layer_test_cube.stl(1)
G0 F3600 X134 Y169 Z0.3
;TYPE:WALL-INNER
G1 F2400 E0
G1 F1500 X116 Y169 E0.89802
G1 X116 Y151 E1.79605
G1 X134 Y151 E2.69407
G1 X134 Y169 E3.59209
G0 F3600 X134.4 Y169.4
G1 F1500 X115.6 Y169.4 E4.53003
G1 X115.6 Y150.6 E5.46797
G1 X134.4 Y150.6 E6.4059
G1 X134.4 Y169.4 E7.34384
G0 F3600 X134.8 Y169.8
;TYPE:WALL-OUTER
G1 F1500 X115.2 Y169.8 E8.32169
G1 X115.2 Y150.2 E9.29953
G1 X134.8 Y150.2 E10.27738
G1 X134.8 Y169.8 E11.25523
G0 F3600 X134.6 Y169.8
G0 X134.33 Y169.33
G0 X133.61 Y168.61
;TYPE:SKIN
G1 F1500 X116.39 Y168.61 E12.11434
G1 X116.39 Y151.39 E12.97345
G1 X133.61 Y151.39 E13.83256
G1 X133.61 Y168.61 E14.69167
G0 F3600 X133.43 Y152.097
G1 F1500 X132.901 Y151.569 E14.72896
G0 F3600 X132.335 Y151.569
G1 F1500 X133.43 Y152.663 E14.80618
G0 F3600 X133.43 Y153.229
G1 F1500 X131.77 Y151.569 E14.9233
G0 F3600 X131.204 Y151.569
G1 F1500 X133.43 Y153.795 E15.08036
G0 F3600 X133.43 Y154.36
G1 F1500 X130.638 Y151.569 E15.27731
G0 F3600 X130.072 Y151.569
G1 F1500 X133.43 Y154.926 E15.5142
G0 F3600 X133.43 Y155.492
G1 F1500 X129.507 Y151.569 E15.79099
G0 F3600 X128.941 Y151.569
G1 F1500 X133.43 Y156.057 E16.10768
G0 F3600 X133.43 Y156.623
G1 F1500 X128.375 Y151.569 E16.4643
G0 F3600 X127.81 Y151.569
G1 F1500 X133.43 Y157.189 E16.86082
G0 F3600 X133.43 Y157.754
G1 F1500 X127.244 Y151.569 E17.29724
G0 F3600 X126.678 Y151.569
G1 F1500 X133.43 Y158.32 E17.7736
G0 F3600 X133.43 Y158.886
G1 F1500 X126.113 Y151.569 E18.28985
G0 F3600 X125.547 Y151.569
G1 F1500 X133.43 Y159.451 E18.846
G0 F3600 X133.43 Y160.017
G1 F1500 X124.981 Y151.569 E19.44209
G0 F3600 X124.416 Y151.569
G1 F1500 X133.43 Y160.583 E20.07808
G0 F3600 X133.43 Y161.148
G1 F1500 X123.85 Y151.569 E20.75396
G0 F3600 X123.284 Y151.569
G1 F1500 X133.43 Y161.714 E21.46978
G0 F3600 X133.43 Y162.28
G1 F1500 X122.719 Y151.569 E22.2255
G0 F3600 X122.153 Y151.569
G1 F1500 X133.43 Y162.845 E23.02112
G0 F3600 X133.43 Y163.411
G1 F1500 X121.587 Y151.569 E23.85667
G0 F3600 X121.022 Y151.569
G1 F1500 X133.43 Y163.977 E24.73212
G0 F3600 X133.43 Y164.543
G1 F1500 X120.456 Y151.569 E25.64751
G0 F3600 X119.89 Y151.569
G1 F1500 X133.43 Y165.108 E26.60279
G0 F3600 X133.43 Y165.674
G1 F1500 X119.324 Y151.569 E27.59801
G0 F3600 X118.759 Y151.569
G1 F1500 X133.43 Y166.24 E28.63313
G0 F3600 X133.43 Y166.805
G1 F1500 X118.193 Y151.569 E29.70815
G0 F3600 X117.627 Y151.569
G1 F1500 X133.43 Y167.371 E30.8231
G0 F3600 X133.43 Y167.937
G1 F1500 X117.062 Y151.569 E31.97795
G0 F3600 X116.57 Y151.642
G1 F1500 X133.356 Y168.429 E33.16233
G0 F3600 X132.791 Y168.429
G1 F1500 X116.57 Y152.208 E34.30681
G0 F3600 X116.57 Y152.773
G1 F1500 X132.225 Y168.429 E35.41139
G0 F3600 X131.659 Y168.429
G1 F1500 X116.57 Y153.339 E36.47604
G0 F3600 X116.57 Y153.905
G1 F1500 X131.094 Y168.429 E37.50078
G0 F3600 X130.528 Y168.429
G1 F1500 X116.57 Y154.471 E38.48559
G0 F3600 X116.57 Y155.036
G1 F1500 X129.962 Y168.429 E39.43051
G0 F3600 X129.397 Y168.429
G1 F1500 X116.57 Y155.602 E40.33552
G0 F3600 X116.57 Y156.168
G1 F1500 X128.831 Y168.429 E41.2006
G0 F3600 X128.265 Y168.429
G1 F1500 X116.57 Y156.733 E42.02578
G0 F3600 X116.57 Y157.299
G1 F1500 X127.699 Y168.429 E42.81103
G0 F3600 X127.134 Y168.429
G1 F1500 X116.57 Y157.865 E43.55638
G0 F3600 X116.57 Y158.43
G1 F1500 X126.568 Y168.429 E44.26182
G0 F3600 X126.002 Y168.429
G1 F1500 X116.57 Y158.996 E44.92734
G0 F3600 X116.57 Y159.562
G1 F1500 X125.437 Y168.429 E45.55295
G0 F3600 X124.871 Y168.429
G1 F1500 X116.57 Y160.127 E46.13867
G0 F3600 X116.57 Y160.693
G1 F1500 X124.305 Y168.429 E46.68445
G0 F3600 X123.74 Y168.429
G1 F1500 X116.57 Y161.259 E47.19033
G0 F3600 X116.57 Y161.824
G1 F1500 X123.174 Y168.429 E47.65631
G0 F3600 X122.608 Y168.429
G1 F1500 X116.57 Y162.39 E48.08236
G0 F3600 X116.57 Y162.956
G1 F1500 X122.043 Y168.429 E48.46851
G0 F3600 X121.477 Y168.429
G1 F1500 X116.57 Y163.521 E48.81476
G0 F3600 X116.57 Y164.087
G1 F1500 X120.911 Y168.429 E49.12108
G0 F3600 X120.346 Y168.429
G1 F1500 X116.57 Y164.653 E49.3875
G0 F3600 X116.57 Y165.219
G1 F1500 X119.78 Y168.429 E49.61398
G0 F3600 X119.214 Y168.429
G1 F1500 X116.57 Y165.784 E49.80056
G0 F3600 X116.57 Y166.35
G1 F1500 X118.648 Y168.429 E49.94721
G0 F3600 X118.083 Y168.429
G1 F1500 X116.57 Y166.916 E50.05396
G0 F3600 X116.57 Y167.481
G1 F1500 X117.517 Y168.429 E50.12081
G0 F3600 X116.951 Y168.429
M104 S260
G1 F1500 X116.57 Y168.047 E50.14773
;MESH:NONMESH
G0 F600 X116.57 Y168.047 Z0.4
G0 F3600 X134 Y169
;TIME_ELAPSED:47.319546
;LAYER:1
;TYPE:WALL-INNER
;MESH:20x20x1mm_first_layer_test_cube.stl(1)
```
</details>

This should start printing only at 260 even if the second extruder is active.

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save Project. For big files you may need to use WeTransfer or similar file sharing sites. G-code files are not project files!)

[project file](https://drive.google.com/file/d/1IdjCKmA9OuGIxmotAXJCMJvRNE-kYv2u/view?usp=sharing)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

nallath, 2020-11-05:
I've just tried this with 4.8 and it doesn't reproduce. It uses 260 correctly. Could you try if it also works for you with the beta?

matthuszagh, 2020-11-17:
Sorry it's taken me so long to respond. I had a bit of difficulty getting 4.8 installed with the new dependencies. In any event, I agree with you about that project file. However, I'm noticing the same effect in a slightly more complicated project file. I've attached it. Mind taking a look?

[project file](https://drive.google.com/file/d/19CGomsZays-hHL1YLCuQzXI-CtJkegoe/view?usp=sharing)

Ghostkeeper, 2020-11-19:
Hmm, I'm still not seeing the same result :( 

Here's the start g-code I'm getting from your latest project file:

```
;FLAVOR:Marlin
;TIME:3332
;Filament used: 0.676804m, 0.779239m
;Layer height: 0.1
;MINX:0
;MINY:0
;MINZ:0.3
;MAXX:134.8
;MAXY:169.8
;MAXZ:16
;Generated with Cura_SteamEngine master
T0
M82 ;absolute extrusion mode
G28
M104 S150
M190 S50.0
G90
G92 E0
M109 S260.0
G92 E0
G92 E0
G1 F2400 E-6.5
;LAYER_COUNT:158
;LAYER:0
M107
;MESH:Multi Extruder Calibration Cube Body1.stl
G0 F3600 X132.422 Y151.051 Z0.3
;TYPE:WALL-INNER
G1 F2400 E0
G1 F1500 X132.807 Y151.176 E0.02019
G1 X133.169 Y151.385 E0.04105
```

The M109 there is the Printing Temperature Initial Layer of the first extruder (T0).
It only mentions lower temperatures for the second extruder starting on layer 33 when it's pre-heating the second extruder just before it needs to go.

I'm wondering a bit where the S150 is coming from. None of the temperature settings are set to 150.

matthuszagh, 2020-11-22:
Very strange, I'm getting the M109 line as `M109 S220.0`. Is there something else that could be influencing this?

matthuszagh, 2020-11-26:
Nevermind, I'm now getting the correct S260. I'm not sure what changed as I'm using the same version. I'll reopen if I see this again. Thanks for the help!

closed 2020-11-26 by matthuszagh