Ultimaker/Cura #8695: Specify different patterns for top and bottom layers instead of a single pattern for both

StarNamer, 2020-11-04:
**Is your feature request related to a problem? Please describe.**
If I want to print the top layer as a concentric pattern, for example, I am forced to also print the bottom layer in a concentric pattern. Currently, the concentric pattern is printed from the inside out, which means that for an unsupported bottom layer, e.g. the cabin roof on a Benchy, the printer will attempt to print in mid-air.

**Describe the solution you'd like**
Issue [#7811](https://github.com/Ultimaker/Cura/issues/7811) suggest printing concentric patterns from outside in which might help, but a better alternative might be to separate the selection of top/bottom patterns and also separate 'concentric' in 'concentric(inwards)' and 'concentric(outwards)'.

**Describe alternatives you've considered**
Obviously, not using concentric is one option, but this limits the final appearance.
Also tree, or other, supports could be used, but it may not be possible to remove these easily and, in any event, they often slightly impair the surface layer.

**Affected users and/or printers**
Any user who wants to use different patterns for top and bottom layers. All printers.

**Additional context**
Noticed while producing a test print from 4.8.0-beta.
[image]

konskarm, 2020-11-05:
Hi @StarNamer. What you are looking for is the `Top Surface Skin Pattern`, which changes the top pattern independently from the bottom. In order to make it visible, you need to have `Top Surface Skin Layers > 0`. With that, you will be able to change some top layers to have the pattern you need, while the rest of the top layers and all the bottom layers will still use the `Top/Bottom Pattern`. Finally, you can change the `Bottom Pattern Initial Layer` if you want to also modify specifically the pattern on the bottom of the print in the initial layer.

Regarding the inward and outward, even if you have your top pattern as concentric outwards, the printer will still attempt to print lines in mid-air, so the concentric(inwards) and concentric(outwards) would make no difference.

Ghostkeeper, 2020-11-10:
Fixed in version 2.7 through the introduction of Top Surface Skin.

closed 2020-11-10 by Ghostkeeper

StarNamer, 2020-11-11:
@Ghostkeeper Thanks. I'll try that.