Ultimaker/Cura #8696: Could the setting label be displayed in a foreign language in the slice error message?

joonghochoe, 2020-11-05:
**Is your feature request related to a problem? Please describe.**
(A clear and concise description of what the problem is. Ex. I'm always frustrated when [...])
If I change to wrong slice setting value in foreign language cura, displayed slice error message. Then error message description is foreign language, but setting label is English.
[image]

[image]

**Describe the solution you'd like**
(A clear and concise description of what you want to happen. If possible, describe why you think this is a good solution.)
I want the setting label to be displayed in a foreign language normally. Foreign language setting label is not applied to "error_labels" in "_onStartSliceCompleted" of "CuraEngineBackend.py".

**Describe alternatives you've considered**
(A clear and concise description of any alternative solutions or features you've considered. Again, if possible, think about why these alternatives are not working out.)

**Affected users and/or printers**
(Who do you think will benefit from this? Is everyone going to benefit from these changes? Or specific kinds of users?)

**Additional context**
(Add any other context or screenshots about the feature request here.)

nallath, 2020-11-05:
This is a bug, since it should have been translated.

konskarm, 2020-11-05:
We suspect that this has the same origin as https://github.com/Ultimaker/Cura/issues/8548

fvrmr, 2022-02-01:
We are constantly working on making Ultimaker Cura better. Our community is a big part of that by requesting new features and reporting bugs. 

To be able to focus on the most requested and needed features and bug fixes from our community and from Ultimaker, we have decided to remove this bug from our backlog. We believe we will not work on this anytime soon and will therefore defer this issue.

Since Cura is open source, you or anyone else is more than welcome to work on this issue and create a pull request yourself.

closed 2023-06-24 by github-actions[bot]