Ultimaker/Cura #8697: [4.8 beta] Material vs profile settings conflict needs to notify the user.

printingotb, 2020-11-05:
**Is your feature request related to a problem? Please describe.**
When I set an absolute value for temp, retract, fan etc in my settings profile, changing material type does nothing.
This is confusing for users as there is no notification that no values changed with the new material.

**Describe the solution you'd like**
Make the user aware of conflicting settings with something like this:
[image]

Perhaps also have a button to the right of each setting to chose what conflict(s) to solve. And the bottom one saying "override all"
That way there also won't be any confusion as to what overrides what since you can immediately see what value changes.
(I think many new users aren't sure what happens with the settings profiles Discard or Keep option. As in, does "keep" means saving the settings on the active profile or moving the changes to the new profile).

**Describe alternatives you've considered**
Display the conflicting settings with a yellow background warning.

**Affected users and/or printers**
Everyone using material profiles. (I don't for various reasons, this being one of them).

Ghostkeeper, 2020-11-10:
I see two problems with this feature.

Material profiles don't just override the 6 or so settings you can see in the material configuration screen. Most of them override some ~20 settings. Many of them are hidden in Cura's interface, but these material profiles may actually override every per-extruder setting in Cura. This isn't necessarily a problem for this feature, but will make the window appear much more complex to the user.

Secondly, you will then get a different result if you first change the custom profile and then the material, than if you were to first change the material and then the custom profile. I think it shouldn't matter to a user in what order they change their profiles. This could lead to a lot of confusion.

Note that this is also a problem for normal quality profiles and intent profiles. Only for unsaved changes will it currently ask. The custom profile is treated the same way as a built-in quality profile in this regard. The custom profile is only available if you're switching to a material of the same Material Type. If the material has the same material type, it is assumed that the same quality profiles, intent profiles and custom profiles still apply. So if your custom profile worked with the original material it should also work for the new material.

I guess the workaround is to not override the temperature in the custom profile. You'd want that temperature override to apply only to some of the material profiles? It's probably better to change the temperature of the material profile then.

printingotb, 2020-11-11:
Hmm, makes sense.

I do it the other way around, I ignore materials and change settings instead, either temporary or by saving a new profile it it's a repeat print. Two prints with the the same material doesn't always need the same temp. It's more convenient to change settings in a settings profile than going into the material menu.

I also don't want the material to limit what quality types are available. If I base my settings profiles of the normal quality type, they are visible for all available materials as fas as I can tell though.

But the point of this issue was to make it easier for the user to understand what's happening when changing materials. I have no better idea at the moment.

closed 2020-11-11 by printingotb

Ghostkeeper, 2020-11-17:
We've seen the UX issues with this dialogue in general more often. People don't understand what it shows. It's not visual enough, too much like a spreadsheet. I hope we'll see a more thorough rework of how that dialogue works and how it looks.