Ultimaker/Cura #8699: Max feedrate of 100mm/s for Tronxy X5SA

onfoot, 2020-11-05:
**Application version**
Any recent one

**Platform**
Any

**Printer**
Tronxy X5SA

**Reproduction steps**
1. Set up an X5SA profile
2. Change travel speed to 150mm/s.

**Actual results**
Travel speed exceeds profile-defined max of 100mm/s.

**Expected results**
This printer is perfectly capable of much faster movement in X and Y. Even if not for printing as this may come down to material or hotend used, but for travel it may be beneficial for reducing stringing or blob forming.

I'm wondering where this limit came from, as for example the creality base profile has set max feedrate to 500mm/s, and this one is inherited by bed-moving printers.

Could the limit be at least overridden with higher values for X5 series specifically or the base have it bumped up? Or was there any evidence that going more than 100mm/s can be harmful for this printer?

nallath, 2020-11-05:
Because whoever wrote the profile (@64bittuning ) set it up like that. I don't know why they did that. But since it's not an Ultimaker printer, we don't test the values that are provided.

The current profile states the following: 
"machine_max_feedrate_x": { "value": 100 },
"machine_max_feedrate_y": { "value": 100 },

If you feel that the values should be different, by all means make a pull request.

64bittuning, 2020-11-05:
The profile is setup in that manner because the printer is limited to those speeds when the firmware is setup as stock.

Had it been setup with a higher limit, someone other than yourself would be here complaining that the machine never reaches the settings they put into the slicer.

I will not be changing the profile.
You may edit it on your end if you'd like.

maciej-rutkowski-intive, 2020-11-05:
Thanks for the answers!

Honestly I've never seen my printer work with the stock firmware or settings because I got mine heavily discounted due to missing crucial parts (no board, display, psu, or heatbed), so it's more of a custom printer with X5SA frame at this point. Maybe it's more appropriate for me to use a custom profile instead of fiddling with the stock ones, even if locally.

onfoot, 2020-11-05:
Thanks for the answers!

Honestly I've never seen my printer work with the stock firmware or settings because I got mine heavily discounted due to missing crucial parts (no board, display, psu, or heatbed), so it's more of a custom printer with X5SA frame at this point. Maybe it's more appropriate for me to use a custom profile instead of fiddling with the stock ones, even if locally.

closed 2020-11-05 by onfoot