Ultimaker/Cura #8700: Cura [4.7.1] and [4.8] unusable slow/lagging if connected with internet over LAN,

jbernuber, 2020-11-05:
**4.7.1 and 4.8**
(The version of the application this issue occurs with.)

**Win 10 Pro 64bit**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Ender 3 V2, but doesnt matter which printer i add**
(Which printer was selected in Cura?)

**Reproduction steps**
1. Booting up the PC
2. It is connect to the internet with an Ethernet Cable
3. starting Cura
4. Network -> Adapter Settings  (the old Win 7 style one)
5. rightclick -> Disable  Ethernet Adapter
6. Cura is responsiv now
7. Enabeling Ethernet Adapter 
8. Cura is super slow again

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 
[image]

**Cura Super Slow**
(What happens after the above steps have been followed.)

**The Ethernet Connection should have no impact on the performance of a programm #2020**
(What should happen after the above steps have been followed.)

**Additional information**
Maybe the Network plugin is responsible for this, regardless, it should not happen!!!
Sorry for spelling, non native english...

fieldOfView, 2020-11-05:
Could you try going to Marketplace -> Installed and disable the "UM3 Network" plugin? Then restart Cura.

jbernuber, 2020-11-05:
> Could you try going to Marketplace -> Installed and disable the "UM3 Network" plugin? Then restart Cura.

Did not make a difference for me, however connecting via WLAN did work, on the same computer...
What black magic is that?
[image]
[image]

fieldOfView, 2020-11-10:
Do you have the OctoPrint Connection plugin installed?

Ghostkeeper, 2020-11-10:
> What black magic is that? Can i smell spaghetti?

UM3NetworkPrinting is a plug-in that searches the LAN and (if you're logged in) your Ultimaker account for Ultimaker printers. fieldOfView is trying to narrow down what is causing the issue by making you disable things that connect to the internet. UM3NetworkPrinting and OctoPrint in particular will try to find printers on the network which is kind of an unusual operation. If antivirus software is suspending the Cura process while it's investigating the network scan, then that could explain why it's slowing down for you but not for anyone else.

Yannickbru, 2020-11-11:
Hello,
I have the same thing at home.
Here is my configuration:

WIN 10 PRO 2004
I5-3570 RAM:16GB 

I tried with two different network cards

Atheros AR8161 motherboard integrated and Realtek PCI GBE Family controller

with kind regards

jbernuber, 2020-11-13:
> Do you have the OctoPrint Connection plugin installed?

Hi, so i made a fresh install of Cura without adjusting anything, exept disabling the "UM3 Plugin".
no Octoprint etc... Apart from that: I expect Cura to work WITH not against Octoprint if in the future I was to choose it.

System: Win 10 Pro 64 20H2
i5 9600k
32Gb Ram
RTX 2070

if you need anything else, feel free to contact me

Ghostkeeper, 2020-11-17:
Hmm, it's a mystery to me as well.

One more idea. Do you happen to have a network drive mapped to the `X:` driveletter?

jbernuber, 2020-11-17:
Yes I do! Why would that matter?

Ghostkeeper, 2020-11-17:
Ooooh, this is a fun one then. We're hitting this old issue: https://github.com/Ultimaker/Cura/issues/1602

Our Windows build server is building Cura on the `X:` drive. This used to be the `D:` drive but we changed that to `X:` because of this bug.

So a detail of Python is that each module gets the property `__file__`, which is the path to where the .py file is located. However on Windows there are no .py files. They all get packaged into Cura.exe. During this packaging, our build system freezes the file names as they were on the build server. So for instance the `CuraApplication.py` module still gets `__file__ = X:\build\4.8\Cura\cura\CuraApplication.py` or something like that.

Nothing should really access any `__file__` of a packaged Python file. It doesn't work on Windows, and would indeed cause Cura to try and access the `X:` drive then. I'm pretty sure that this doesn't happen in our own code base. We can access `__file__` only in the plug-in modules (which are not packaged in the .exe) or when not `frozen` (an environment variable determining whether you're running from source or not). However we don't know what CPython does here and perhaps some of Cura's dependencies do access the file system relative to a `__file__` because Python dependencies are packaged inside the Cura.exe too.

I'll close this as a duplicate of https://github.com/Ultimaker/Cura/issues/1602 then, sorry. I'll add a note to that issue that the effect extends to slowness if there is a network drive with the `X:` driveletter.
I'm wondering if this could result in a security issue. I don't think it's likely, but if one of our dependencies is accessing a random network drive and tries to execute code from there (perhaps because it is expecting a relatively-positioned Python module) that could be a problem.

closed 2020-11-17 by Ghostkeeper