Ultimaker/Cura #8701: Tree support completely omitted for one side of symmetric model

StarNamer, 2020-11-06:
I got this the first time I tried slicing the model. Moving it slightly in the Y direction and reslicing gave the expected behaviour of supports in both ducts, but since the is 4.8.0-Beta I thought I'd log it in case there's something that can be done to avoid completely omitting a tree support.

**Application version**
4.8.0-Beta

**Platform**
Windows 10 Insider Preview Build 20246

**Printer**
Ender-3

**Reproduction steps**
Sliced symmetric model

**Screenshot(s)**
Duct on the right has an internal support, the one on the left doesn't.
[image]
[image]

**Actual results**
Cura inserts tree supports for one part of model, but not for the identical part on the other side, as shown in screenshots.

**Expected results**
Roughly similar supports on both sides (within the randomness of tree supports).

**Project file**
[CE3_6fd0e370-37f0-4c17-8ae1-05368710ba89.zip](https://github.com/Ultimaker/Cura/files/5498232/CE3_6fd0e370-37f0-4c17-8ae1-05368710ba89.zip)

**Log file**
[cura.zip](https://github.com/Ultimaker/Cura/files/5498243/cura.zip)

**Additional information**
I'm printing it anyway, since I didn't notice until I'd started. I will update on success or otherwise.

Ghostkeeper, 2020-11-10:
Hmm, this doesn't seem to reproduce any more in the stable release. It's creating support on both sides now.

[image]

We didn't make a fix aimed at this in particular in between. We've changed the Maximum Resolution setting for your printer since the beta, but changing that back to 0.05mm doesn't reproduce the issue either.

Is it still occurring in 4.8.0 stable for you?

StarNamer, 2020-11-11:
Model printed successfully. Further slicing attempt withs 4.8.0-Beta could not reproduce the problem. I'll try a few times with 4.8.0 later, but suspect it was simply an extreme random case for tree supports. Next time I'll check and re-slice if I see this.

Ghostkeeper, 2020-11-11:
Cura's slices are expected to be deterministic by the way, unless you're using random Z seam placement or fuzzy skin (which both use explicit random numbers). The placement of the tree's branches is according to a grid. Before 4.7 there was a bug with tree support that made it nondeterministic (by reading uninitialised memory) but since then we don't know of any randomness.

GregValiant, 2024-11-20:
Is this still an issue in current versions of Cura (5.8.0 and up)? Can this be closed?

StarNamer, 2024-11-20:
I haven't had any further examples since the 4.8.0 beta. Closing.

closed 2024-11-20 by StarNamer