Ultimaker/Cura #8702: Command-V (Paste) does not work on the file-name input dialogue at save on Mac

gitcnd, 2020-11-06:
**Application version**
4.7.1

**Platform**
Mac

**Reproduction steps**
1. Click SAVE TO FILE
2. Try to paste the name for the file (not using the mouse)

Ghostkeeper, 2020-11-10:
Indeed this is a bug in the Qt framework that we command to create the file save dialogue: https://bugreports.qt.io/browse/QTBUG-17291

It was fixed in 5.12.0, then [regressed](https://bugreports.qt.io/browse/QTBUG-85679) in 5.12.8 and was fixed again in 5.15.1. So basically to fix this we'd need to update to Qt 5.15.1. We can't really do that though since that would drop support for MacOS 10.10, 10.11 and 10.12. We may consider dropping 10.10, but beyond that the MacOS versions are still in use by a significant fraction of our userbase.

nallath, 2021-06-10:
We've updated to Qt 5.15, so this should be fixed for 4.9.1 and up.

closed 2021-06-10 by nallath