Ultimaker/Cura #8703: [4.8 beta] M201 and M202 (acceleration) added at the end of gcode with very high values

printingotb, 2020-11-06:
**Application version**
4.8 beta

**Printer**
Custom FFF printer, with repetier gcode flavour

**Reproduction steps**
1. Open project file
2. Slice and save Gcode

**Screenshot(s)**
[image]

**Actual results**
Cura adds M201 and M202 commands at the very end, before the custom end gcode, with a value of X 4000 and Y 4000.
Is it trying to restore the "default" acc value?
I can't find any values with 4000 anywhere in my acceleration settings or start gcode, perhaps they are defined in the printer profile?
Doesn't this mean that the following print will be started with an acceleration of 4000 if it has acceleration control turned off and there are no acc commands in the start gcode? The first M201 value is 500 and the first M202 value is 1500.

**Expected results**
Shouldn't add such high values.

**Project file**
[M201 and M202 4000.zip](https://github.com/Ultimaker/Cura/files/5500205/M201.and.M202.4000.zip)

**Additional information**

Ghostkeeper, 2020-11-10:
Indeed, it's setting them back to the Default Acceleration as defined in the machine settings for your printer. After completing the print, if Cura took over acceleration control, it's setting it back to what Cura thinks is the default for your printer, so that prints made with other slicers (or if acceleration control is disabled in Cura later) will print with proper acceleration, rather than the acceleration of whatever Cura decided to print last.

The Default Acceleration setting is a hidden setting. If you download the Printer Settings plug-in in the Marketplace by fieldOfView, you'll be able to change this setting too.

This is expected behaviour, so I'll close this issue.

closed 2020-11-10 by Ghostkeeper

printingotb, 2020-11-11:
Here I was thinking that 4000 was too high for most hobby grade printers (this is for the custom fff profile) but then I noticed that the default "Max Speed" was set to the speed of light. Did you choose that specific number on purpose? :sweat_smile: 
Not that it's modifying the gcode with this value.

Is the thinking that all printers have safe max values set in the firmware that can't be overridden? A lot of printers would loose steps with an acceleration of 4000. Default X/Y jerk of 20 is around the limit for some. When I create settings profiles supposed to work with many different printers, the highest values I use are 1500/15 for travel moves and lower for the rest.

Ghostkeeper, 2020-11-17:
We have the philosophy that our hard limits of the settings should only be imposed by universal laws of physics or mathematical impossibilities. So the maximum speed is indeed set to light speed :grin: 

I think the Maximum Acceleration values in fdmprinter have been taken from the configuration of the Ultimaker 2 when those settings were added in 2016. The Ultimaker 2 was pretty high-end acceleration-wise since it's a Bowden printer.

fieldOfView, 2020-11-17:
> We have the philosophy that our hard limits of the settings should only be imposed by universal laws of physics or mathematical impossibilities.

Except when it comes to temperatures, at which point you set a fairly arbitrary maximum. But that is offtopic.

Ghostkeeper, 2020-11-17:
True. That's not very consistent. I think there were some safety discussions involved there as to what happens when a user just instructs the printer to print at 9999 degrees C. It was upped in January when people were running into the limit with high-temperature plastics. Luckily not a whole lot of people run into the issue for now, but that's definitely worth reconsidering.