Ultimaker/Cura #8704: Z Seam Alignment issue

scubamario, 2020-11-06:
**Application version**
Cura 4.8 Beta

**Platform**
Window 10

**Printer**
Ender 5 Plus

**Reproduction steps**
1. I was thinking of printing the little Among Us characters from Thingiverse for my Son (https://www.thingiverse.com/thing:4615103) and decided to give it a try with the new Cura 4.8. I was using the same Z Seam profile (Z Seam Alignment:User Specified / Z Seam Position:Back / S Seam X:175.0 / Z Seam Y:350) from the previous Cura 4.7.1 and when the print was done, I noticed in the back of the print, the Z Seam was diagonal on both parts.
2. I decided to print the same file but with Cura 4.7.1 and the seam was perfect.

**Screenshot(s)**
[image]
[image]
[image]
[image]

**Actual results**
Better Z Seam Alignment with version of Cura 4.7.1

**Expected results**
Align Z Seam and not random with Cura 4.8

**Project file**
N/A

**Log file**
Sorry, I tried to find it but no luck. Instructions are not very clear.

**Additional information**
N/A

konskarm, 2020-11-06:
I believe this is related to an issue we had, where the Z seam was being placed halfway through a straight wall. This issue is fixed in https://github.com/Ultimaker/CuraEngine/pull/1344, so it should be there in 4.8 stable. Nevertheless, could you please provide a project (.3mf) file that displays this issue? I cannot reproduce it locally.

To generate a project file go to "File"->"Save Project...". You need to zip it before uploading it in GitHub.

scubamario, 2020-11-07:
[Among_us_helmet.zip](https://github.com/Ultimaker/Cura/files/5504203/Among_us_helmet.zip)

konskarm, 2020-11-09:
I think I see the problem. By default, the printer has set the `Z Seam Alignment` to `User Specified`, and the `Z Seam Position` is set to `(175,350)`. 

[image]

This just ensures that the Z seam will be somewhere in the back, but since the user-specified z seam position is so far from the model itself (due to the Y Seam position), it cannot force the Z Seam to be aligned in the Z direction. 

I can see two ways to fix this (**Note:** I enabled coasting so that the Seam is visible on the model. It won't actually look like that when printed.)

- Select a `Z Seam position` as `(175,175)`, so that the Y is close to the model and the starting point of each layer will be selected close to that Y
[image]

- Change the `Z Seam Alignment` to `Sharpest Corner` and let the `Seam Corner Preference` as `Hide Seam`. 
**Back**:
[image]

**Front**:
[image]

In any case, this isn't really a bug, but rather a settings issue.

scubamario, 2020-11-09:
Hi,

The parameters of the version 4.7.1 and 4.8 was exactly the same for the Z Seam position (X & Y) so they should be identical on both prints, right???

konskarm, 2020-11-10:
It should indeed. When I open the file in 4.7 and slice it without changing any settings, I get exactly the same result to the 4.8-beta (again coasting enabled for visualization)

[image]

Am I missing something here?

Ghostkeeper, 2020-11-10:
It's placing the seam across the diagonal since the flat back face is comprised of 2 triangles with an edge in the middle there. The edge is closest to the seam position so the seam gets placed there.

I'd expect the edge to be removed by the simplification algorithm though. That would be the actual bug then?

konskarm, 2020-11-11:
That is indeed true. It should be removed by the new simplify. I'll look into why it doesn't.

konskarm, 2020-11-11:
After closer inspection, it turns out that the three vertexes in the back are not exactly colinear. As an example, layer 50 in the back looks like this in a scatter plot:
[image]

The 3 highlighted points actually have a small angle between them, so the middle point isn't removed by simplify. Now since its an angle barely greater than 180 degrees, it is always selected by the "Hide Z Seam" Seam Corner Preference leading to that diagonal.

Changing the Seam Corner Preference to anything other than "Hide Z Seam" should not create the diagonal. As an example, having "Smart Hiding":  
[image]

All-in-all, I would say this is expected behavior, although I am unsure how desired it may be.

I am still unsure though how you got different results for 4.7.1 and 4.8 without changing any settings.

Ghostkeeper, 2020-11-17:
Yes, I agree. This is expected behaviour then. It's not removed by simplification since the segments are longer than the Maximum Resolution. It's being selected as an option since the corner is actually the inner corner closest to the desired seam position.

Smart Hiding won't change that since the alignment is user-specified, so it must always be the closest seam position that matches the corner preference. Fixing the model so that the front is actually flat would resolve the problem, or selecting "shortest" or "sharpest corner" as seam location.

closed 2020-11-17 by Ghostkeeper