Ultimaker/Cura #8706: New Materials - Missing Profiles

Kaicura, 2020-11-06:
<New Materials  like SPLA Missing 
The are no Profils  like Standart arrivel  to selekt when new Material  Type like SPLA  is generatet.
Thank you for using Cura!
--> 
[image]

**CURA 4.8.0 Beta**

**WIn 7**

**Printer**
Artillery Sidewinder X1
**Reproduction steps**
1. (Make a New Material Type like SPLA.)
2. (Aktivate New material type
[image]
The Window whit Material colored Orange 

*

jellespijker, 2020-11-07:
It is not possible for us to create and update the profiles for all the new materials out there on the third-party printers.
Looking at the quality profiles of the artillery printers I see they're defined for **TPU, PLA, PETG, ABS** (https://github.com/Ultimaker/Cura/tree/master/resources/quality/artillery)

There are two possible roads you can take to solve this problem, either create a new quality profile for **SPLA** or you can change your material profile and base it on **PLA**. I assume that the **SPLA** you mention stands for **silk PLA**, which is still a **polyactic acid** based polymer but has some additional additives to give it a silk like shine.

closed 2020-11-07 by jellespijker

Kaicura, 2020-12-17:
Nein , SPLA ist einMIX aus 50% PETG, 30%PLA und20% TPU