Ultimaker/Cura #8710: Include eSteps in the Material profile

amantegazzi, 2020-11-07:
**Is your feature request related to a problem? Please describe.**
Before I use new materials, I calibrate the extruder. I keep a note so I can set the correct eSteps for each material.

**Describe the solution you'd like**
Having eSteps in the material profile, would save me some time and less error prone.

**Describe alternatives you've considered**
Currently setting eSteps manually.

**Affected users and/or printers**
Everyone

**Additional context**
none

Asterchades, 2020-11-08:
I mean I get it - there's settings I'd like to see there as well (an additional set of start-up GCode would be great). But isn't the Flow setting sufficient for this in pretty much every situation it's likely to matter?

MrEd88, 2020-11-08:
Well, since E-steps needs to be calibrated for each roll of filament it would be a good idea. Actually, yesterday I asked myself if that was already there in material settings. Currently I have 2 machines, one of them is using klipper as firmware, I am creating custom macros to set Pressure advance and esteps for each of my filaments, so I dont include anything in Gcode startup and just run the macro of the filament before the sliced gcode. My other machine is running Marlin, I guess there is an octoprint plugin to send the Gcode before the sliced one.

Including those settings on filament manager should be a simple task.

Asterchades, 2020-11-08:
There is a "Sidebar Macro" plugin for OctoPrint which can theoretically do that, but obviously that's of little help to people who don't use OctoPrint. I don't believe it can be invoked via GCode like Klipper shortcuts, though, so it requires manual execution.

Even so, that doesn't really explain why having an E-steps setting is required above and beyond the Flow setting. You might get slightly more precision with an E-step setting but the Flow is already permissible down to 3 decimal places so you'd need to be using an extremely high gear ratio for it to even make a mathematically relevant difference, never mind a practically relevant one.

amantegazzi, 2020-11-08:
eSteps and Flow rate are two different parameters and each one has its own reason of being. Personally I use flow settings rarely, surely other people might use it instead of changing eSteps. That said changing the flow rate to achieve good results usually takes a few prints/more time to get "right" and still is subjective to each and everyone's expertise. It is certainly easier to change than eSteps as you can do it on the fly while during printing. In my opinion flow settings should be used to fine tune, having the right eSteps is the first step toward getting the correct extrusion. Just my 2cts.

Asterchades, 2020-11-08:
They're separate but functionally equivalent, even if not identical. Adjusting one is the same as adjusting the other, just on a different scale. To wit, if you're out by 1% on your E-steps you can either adjust the E-step value down by 1% of its current value, or you can adjust the Flow down by 1% of its current value - both will achieve the same results, albeit at different stages in the process (and I can see merit to adjusting both separately, especially if you run multiple printers).

> It is certainly easier to change than eSteps as you can do it on the fly while during printing. 

M220 S<percentage>

You can adjust Flow on the fly as well, either by GCode (Pronterface, OctoPrint, etc) or many printers have it exposed by default via Marlin for adjustment via the LCD menus. You may need to math a bit if your base Flow is not 100% but you already had to math a bit for E-steps so that's neither here nor there.

NB I'm not disagreeing that it should be added. Just pointing out that, at least on the surface, the setting is redundant given what's already exposed can achieve the same outcome.

fieldOfView, 2020-11-08:
This should be possible by using the "Material Settings" plugin (https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/MaterialSettingsPlugin) in combination the "Printer Settings" plugin (https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/PrinterSettingsPlugin). The latter will expose the "Steps per Millimeter (E)" setting in the settings sidebar, and the former will then let you add it to the Materials pane of the preferences.

nallath, 2020-11-09:
As far as I know, flow is indeed the same as Steps per Milimeter. If you look at the M221 g-code, you can see that that is indeed the case: https://marlinfw.org/docs/gcode/M221.html

Anyhow, since there are options to do this with the respective plugins, I think this can be closed.

closed 2020-11-09 by nallath

amantegazzi, 2020-11-10:
The fact that eSteps and Flowrate both result in changing the extruder's stepper motor "speed" is irrelevant to this discussion imo.
I also get that flow is already a material setting and therefore adding eSteps is "technically" redundant.

My point is totally about user experience. Getting the correct flow is more an art than a science imo. It involves to complete a series of printing - measuring - tweak - repeat. Therefore this is a lengthier and time consuming process and has a higher number of variables that contribute to influence result. Calculating the eSteps is by far a faster and simpler process.

That said, the above is my assumption based on my own experience. It would be great if you had stats on what most users do actually prefer to use eSteps or flow when they change material.

Granted I wasn't aware of the plugins, which could be a good solution. @Asterchades made a good point that this could lead to problems, so this is definitely not something simple to do but hey that's why you, smart bunch of dedicated people are here for. I don't have the time nor the expertise to help unfortunately.

Anyway, thanks for taking the time to review and comment. I, for sure will continue to use eSteps for each material/manufacturer combo and use flow only if needed to fine tune results.