Ultimaker/Cura #8711: Bug Cura se reinicia al abrir

Porles, 2020-11-08:
**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save Project. For big files you may need to use WeTransfer or similar file sharing sites. G-code files are not project files!)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

Porles, 2020-11-08:
Tengo el cura para laminar, pero en muchas ocasiones cuando abro el programa Cura, este se abre como recien instalado, sin existir en el mis maquinas configuradas y perfiles añadidos. Es muy frustrante. Cada vez hay que configurar las maquinas de nuevo...

nallath, 2020-11-09:
It seems that the issue template has not been filled in entirely. Please check our guide on https://github.com/Ultimaker/Cura/wiki/Reporting-issues for more information on what information we need and why we need it.

We would also like to ask you to post your bug reports in English.

closed 2020-11-23 by no-response[bot]