Ultimaker/Cura #8712: Creality CR-6 SE Profile not avaliable

BakaAu, 2020-11-08:
**Application version**
4.7.1

**Platform**
windows 10 - gtx 1060

**Printer**
Issue is cant select cr-6 se - i have seen others screen shot it as avaliable and we are using same version but it is not in the list of Creality printers for me

**Reproduction steps**
1. downloaded newest version
2.verified with others printers profile was avaliable for them

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
profile not avaliable

**Expected results**
profile should be listed

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save Project. For big files you may need to use WeTransfer or similar file sharing sites. G-code files are not project files!)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

nallath, 2020-11-09:
I've just checked it in master and it seems to be there. Is it available for you for the 4.8 beta?

DreadKnight, 2020-11-12:
This should to be closed, someone hasn't used the search function. The printer was actually added in 4.8

closed 2020-11-13 by nallath