Ultimaker/Cura #8713: [4.8 Beta] Not slicing when using adaptative layers and model is not touching build plate

MrEd88, 2020-11-08:
**Application version**
4.8 Beta 

**Platform**
Linux Mint 20 - GTX 1050TI

**Printer**
Custom printer

**Reproduction steps**
1. Load model, raise it any distance from bed and enable supports
2. Enable and adjust adaptative layers
3. Slice and review

**Screenshot(s)** 
[image]

**Actual results**
Cura seems to slice in a quarter of a second but result in no slicing at all.
If adaptative layers is disabled it slices normally.
If model is moved to touch the build plate leaving enabled adaptative layers, it slices normally.

**Expected results**
Slice normally 

**Project file**
N/A. Any model with any profile doing the steps mentioned before.

**Additional information**
Did not test in older versions, currently using only the beta works for most of my needs.

konskarm, 2020-11-11:
This is now in our backlog. Devs see CURA-7835.

bucefalo2, 2020-11-12:
exactly same problem i faced today ! thank you !!!

Ghostkeeper, 2020-11-17:
Correction for developers: The actual issue is CURA-6513. We already had this in our backlog.

This issue is a duplicate of https://github.com/Ultimaker/Cura/issues/3183. Please continue discussion there.

closed 2020-11-17 by Ghostkeeper