Ultimaker/Cura #8714: About Draft Shields and Skirts

viper0078, 2020-11-09:
The draft shield is intended to keep the printed material warm.
In addition, the skirt can keep the bed warm by setting the distance between the skirt and the printed material to about 0.7mm, making it difficult for the ABS to come off.

In the case of Cura, it is difficult to achieve these heat-retaining effects when multiple pieces are printed at the same time.
The draft shield has a very large gap and the thermal effect is reduced when the bed is in operation, as in the case of SnapMaker.
[image]

Simplify3D prints a shield or skirt for each printout, and I wish Cura would do the same.
[image]

And I want to specify these flow rates separately. In particular, the skirt only creates one layer, and I want to create a thick layer without being affected by the parameters of InitialLayerFlow.

Please consider it.

konskarm, 2020-11-09:
From the images it seems that this is not a skirt, but a brim, and the brim indeed splits to be per model only. I am not entirely sure why the skirt would need to be split to be per model. However, indeed the draft shield is merged among models and I can understand that it can lead to losing heat-retaining effects. I'll put this up for discussion with the team.

viper0078, 2020-11-09:
I learned the following on the Cura forum.

Creating a skirt may provide a warming effect on the bed, thereby reducing the warping of the ABS. with a SkirtDistance of 0.1 mm, it will not be glued to the printed material either.

The current Cura skirt is less likely to produce the effect described above when multiple prints are printed.

For reference, Simplify3D has also been able to create a shield that follows the shape of the model, as shown below Depending on the shape of the shield, it may have a greater heat retention effect.
[image]

Please consider it.

konskarm, 2020-11-10:
Please create separate tickets for different features next time, as it helps us keep a better track of the discussions on each ticket (for example, the draft shield that follows the shape of the model can easily be missed as it is not a separate thread).

That being said, we decided that we will reject the skirt-splitting feature and we will defer splitting the draft shield, meaning we are open to pull requests that achieve this change.

Regarding the draft shield that follows the shape of the model, we are still considering it. I'll let you know once we have a decision.

konskarm, 2020-11-11:
Team decided to add the "Draft shield follows model" in our backlog. Devs see CURA-7837.

viper0078, 2020-11-11:
Thank you for your consideration. I'm looking forward to seeing the features added.